esphome/feature-requests #290: Dashboard Import of yaml config

mehuman, 2019-06-19:
**Describe the problem you have/What new integration you would like**

When adding a new device to the dashboard it would be great to be able to import a yaml file in order to skip filling out the wizard with redundant board and wifi information.

**Please describe your use case for this integration and alternatives you've tried:**

Adding new devices would be quicker and more streamlined. I manage my configs in a git repo so it would allow me to recover quicker in the case of data loss. It also allows me to stub out a basic template.

I have not tried but read that the dashboard just reads yaml files from a directory and if a file is added/removed the dashboard is updated. The import could be as simple as just uploading the yaml file to that directory.

**Additional context**

OttoWinter, 2019-06-20:
FYI, you can also just copy+paste the files into the esphome config folder.

rradar, 2019-06-21:
To have (the possibility) to skip the wizard would be really handy. :+1:

nagyrobi, 2022-07-07:
As @OttoWinter said, it's possible to just drop any .yaml file esphome config folder with the appropriate config already created manually, the Dashboard will pick it up. No wizard or anything will appear.
Closing this as the request is already supported.