esphome/feature-requests #291: Let esphome declare service in mDNS mode, so that IP and port can be found by other software

IoTServ, 2019-06-22:
**Describe the problem you have/What new integration you would like**

Let esphome declare service in mDNS mode, so that IP and port can be found by other software
**Please describe your use case for this integration and alternatives you've tried:**

Let the service and interface of esphome be discovered by other software, so that other software can find the address and interface of esphome more conveniently, and then better cooperate with other software.
**Additional context**

IoTServ, 2019-06-22:
like declare as _http._tcp.local or _esphome._tcp.local

OttoWinter, 2019-06-22:
Already exists - if you have native API enabled, you have `_esphomelib._tcp.local` (esphomelib for legacy reasons) and otherwise `_http._tcp.local`.

https://github.com/esphome/esphome/blob/dev/esphome/core/util.cpp#L45

closed 2019-06-22 by OttoWinter