esphome/feature-requests #293: Xiaomi Yeelight Smart Dimmer Switch

kuradi, 2019-06-24:
**Describe the problem you have/What new integration you would like**

Xiaomi Yeelight Smart Dimmer Switch
https://www.xiaomitoday.com/xiaomi-yeelight-smart-dimmer-switch-the-intelligent-light-switch/

**Please describe your use case for this integration and alternatives you've tried:**

dim lights, dim tv volume etc etc

**Additional context**

AR64SH, 2019-07-09:
I'd like it too. I've one of this switch. it would be very interesting to have it works with my Yeelight Smart Bulb using esphome

kuradi, 2019-08-07:
maybe this helps a little bit https://github.com/nccchirag/yeelight-ble-rotary-dimmer

minsuke, 2019-08-26:
If this dimmer is available esphome will be a more and more greatest tool, thx to Otto Winter. I respect you

nagyrobi, 2022-07-06:
Repository above has the component to use this.