esphome/feature-requests #294: automation on deep sleep begin

fondberg, 2019-06-24:
**Describe the problem you have/What new integration you would like**

Automation trigger right before deep sleep begins

**Please describe your use case for this integration and alternatives you've tried:**

Be able to turn off any GPIO switches that possibly powers sensors. 

**Additional context**

Possible configuration:
```yaml
deep_sleep:
  id: deep_sleep_1
  run_duration: 2min
  sleep_duration: 5min
  on_wakeup:
    then:
       ....
  on_sleep:
    then:
       ...
```

OttoWinter, 2019-06-24:
`on_sleep:` that _should_ be handled by `on_shutdown` in `esphome:` section. If not, that's a bug.

fondberg, 2019-06-24:
I see. I'll report a bug then because I don't see them working

closed 2019-07-03 by OttoWinter