esphome/feature-requests #296: Possibility to serve time gathered by gps component

rradar, 2019-06-25:
**Describe the problem you have/What new integration you would like**

I like to have the possibility to serve the time gathered by the gps component. For example as a ntp server.

**Please describe your use case for this integration and alternatives you've tried:**

I need to have a valid time source for an offline environment.

As all esp and even raspberries lacking a rtc it would be great to use the (supported) time source from gps which should have good accuracy (stratum 1) and should be even better than rtc's. 

**Additional context**

https://github.com/liebman/ESPNTPServer
https://hackaday.io/project/28013-esp-gps-ntp-server

rradar, 2019-08-26:
ntpd library: https://github.com/jonsl/arduino-ntpd/tree/master/arduino-ntpd

daubman, 2025-03-28:
It would be great to get @RobertJN64's [pull request](https://github.com/esphome/esphome/pull/4902) added for NTP server component - is there anything I can do to help make that happen - it wasn't clear why their pull request was closed?

nagyrobi, 2025-03-28:
Likely because low priority.
You could re-submit it with code changed to be able to be merged to current versions.
Also update the docs PR.