esphome/feature-requests #297: Allow a node to define the time period the node is disconnected before reporting itself as unavailable

vamichael, 2019-06-25:
**Describe the problem you have/What new integration you would like**

Be able to specify in the control a time period a node is disconnected from the ESPHome server before the node reports itself as being unavailable.

From within the yaml, when defining the node have an option that defines the timeout period which determines when a node’s last will and testament reports itself as `unavailable`. I’m assuming this is currently handled with MQTT and the last will and testament feature.  I also assume this is currently triggered after so many failed WiFi attempts (or time length).

The following yaml works, and will report in HA a `sensor.bmw` an incrementing number every second when a WiFi and MQTT session is established.  If the node loses power or exits WiFi range, after a period of around 15 seconds the sensor value changes to “unavailable”.  I would like to shorten this time period.

What I’d like to be able to do is define this timeout period when defining the node with a `timeout` definition.  See below where I define this mythical timeout for 5 seconds.  Perhaps you can do something like this already, but I haven’t discovered it yet.

```
esphome:
  name: car_bmw
  platform: ESP8266
  board: d1_mini
  timeout: 5000ms <——— define timeout
 
wifi:
  ssid: "MySSSID"
  password: "0123456789"
  fast_connect: true
  use_address: 172.16.68.121
  reboot_timeout: 0s
 
  manual_ip:
    static_ip: 172.16.68.121
    gateway: 172.16.68.1
    subnet: 255.255.255.0
 
mqtt:
  discovery: true
  discovery_prefix: homeassistant
  client_id: home-assistant-esphome-bmw
  broker: 172.16.68.62
  username: mqtt
  password: mqttpassword
  reboot_timeout: 0s
   
logger:
ota:
  password: 'pass'
 
binary_sensor:
  - platform: status
    name: "Garage Opened BMW"
    id: garage_opened_bmw
 
sensor:
  - platform: template
    name: "BMW"
    lambda: |-
   
      static int num_cycles = 0;
      static int prev_count = 0;
      int retVal;
      num_cycles += 1;
      bool ha_responded = (id(garage_opened_bmw).state); // Home Assistant set binary_sensor ON
      if (ha_responded) {
        retVal = prev_count;
      }
      else
      {
        retVal = num_cycles;
        prev_count = num_cycles;
      }
      return retVal;
   
    update_interval: 1000ms
```

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

brandond, 2019-06-26:
Home Assistant picks up on the Unavailable state by watching the node's status topic. When the node disconnects from MQTT, the LWT message's "offline" payload gets published to that topic. Since this all occurs on the MQTT server as an automatic action when the client disconnects, there's no real way to configure a specific timeout - other than perhaps altering the client connection timeout in the server config, if your client is simply dropping off the network instead of disconnecting cleanly.

https://esphome.io/components/mqtt.html#mqtt-last-will-birth

OttoWinter, 2019-06-26:
Correct, as @brandond said.

LWT (through which unavailable is set) is published by the broker immediately upon an uncontrolled shutdown. This can not be delayed (though you can increase the keepalive time, but that only handles network lost disconnects).

This *could* be solved in HA through a new option, but

 - This is not a feature request tracker for HA too
 - I doubt it would be accepted, too niche of a use-case

You can use HA's template sensors and such to work around this.

closed 2019-06-26 by OttoWinter

vamichael, 2019-06-27:
> Home Assistant picks up on the Unavailable state by watching the node's status topic. When the node disconnects from MQTT, the LWT message's "offline" payload gets published to that topic by the broker. The delay after which this occurs is configured by the keepalive setting.
> 
> > keepalive (Optional, Time): The time to keep the MQTT socket alive, decreasing this can help with overall stability due to more WiFi traffic with more pings. Defaults to 15 seconds.
> 
> https://esphome.io/components/mqtt.html#configuration-variables
> https://esphome.io/components/mqtt.html#mqtt-last-will-birth

So this might be configured to a lower timeout in the HA configuration of MQTT?