esphome/feature-requests #298: Cookbook entry how to make a buzzer work

rradar, 2019-06-26:
**Describe the problem you have/What new integration you would like**

Many people don't know how to buzzer (including me)

**Please describe your use case for this integration and alternatives you've tried:**

I tried to clap my hands but can't get this automated with my esp

**Additional context**

Would be nice to have not only a single buzzzzzz in the cookbook but also buzzing a little jingle or something :smile_cat:

rradar, 2019-08-13:
Let's start with a sketch for a passive buzzer which finally works :1st_place_medal:

```
output:
  - platform: esp8266_pwm
    pin: D7
    id: 'buzzer'

switch:
  - platform: template
    name: buzzzzzz
    turn_on_action:
      then:
        - output.esp8266_pwm.set_frequency:
            id: buzzer
            frequency: 800Hz
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 120ms
        - output.turn_off: buzzer
        - delay: 200ms
        - output.esp8266_pwm.set_frequency:
            id: buzzer
            frequency: 600Hz
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 120ms
        - output.turn_off: buzzer

```

aghosh0605, 2024-06-02:
> Let's start with a sketch for a passive buzzer which finally works 🥇
> 
> ```
> output:
>   - platform: esp8266_pwm
>     pin: D7
>     id: 'buzzer'
> 
> switch:
>   - platform: template
>     name: buzzzzzz
>     turn_on_action:
>       then:
>         - output.esp8266_pwm.set_frequency:
>             id: buzzer
>             frequency: 800Hz
>         - output.set_level:
>             id: buzzer
>             level: 50%
>         - delay: 120ms
>         - output.turn_off: buzzer
>         - delay: 200ms
>         - output.esp8266_pwm.set_frequency:
>             id: buzzer
>             frequency: 600Hz
>         - output.set_level:
>             id: buzzer
>             level: 50%
>         - delay: 120ms
>         - output.turn_off: buzzer
> ```

Hey, thanks...literally, I was finding some help on how to implement it. I was Scrolling through a lot of forums. Luckily, your answer helped me to understand it.

stedrow, 2024-11-29:
> Let's start with a sketch for a passive buzzer which finally works 🥇
>
> ```
> output:
>   - platform: esp8266_pwm
>     pin: D7
>     id: 'buzzer'
>
> switch:
>   - platform: template
>     name: buzzzzzz [...]

Sorry for the bump on a very old thread but this code example was extremely helpful and got my project working!