esphome/feature-requests #299: Canbus support for esp32

mvturnho, 2019-06-27:
**Describe the problem you have/What new integration you would like**

Support for Canbus to have multiple esp32 devices communicatie sensor activity and initiate actuators and light management.

**Please describe your use case for this integration and alternatives you've tried:**

Wifi is sometimes unavailable or slow. To have a more reliable connection between esp32 device I would like to connect them over canbus.
Mostly it could be done from the automation framework.

Wahen for instance a button is pressed the action is published with a can id for the specific sensor on the canbus.
Other devices that respond to the canbus id for this buttonpress perform actions on lights, relays or actuators by caling automations functions.

**Additional context**

example yaml file (proposal)
```yaml

canbus:
  canid: 666
  frequency: 500k
  rx_pin: 4
  tx_pin: 3
  on_receive: 190
    then:
      - switch.turn_on: switch1 
  on_receive: 191
    then:
      - switch.turn_off: switch1 
    
binary_sensor:
  platform: gpio
  pin: gpio5
  id: sensor1
  on_press:
    then:
      - send_canbus: 190
  on_release:
    then:
      - send_canbus: 191

switch:
  platform: gpio
  pin: gpio7
  id: switch1
```

mvturnho, 2019-07-01:
@OttoWinter There is a candriver in the arduino supportlib for ESP32.
Question is if the freeRTOS is supported or allowed in esphome?

mvturnho, 2019-07-01:
Maybe its best to start with MCP2515. Then ESP8266 is also supported. This device connects to the SPI.

OttoWinter, 2019-07-01:
This falls in the communication channel bucket - the plan is to structure native APIv2 in such a way that it allows communication across many channels (tcp, mesh networking, ...), then that allows for things like API services etc for inter-device communication.

Let's also not mix CAN _bus_ and CAN as an inter-device communication. CAN _bus_ would be one integration (like i2c, spi etc) that only handles the CAN bus communication. For example external sensors that use the CAN bus protocol can hook into it like with i2c.

CAN communication layer would be another thing on top of the bus.

As for freertos: yes, freertos can be used directly on ESP32 (see for example also esp32 camera, uses mutexes and esp-idf apis). But: Unless CAN cannot be implemented on ESP8266, there should be a second software implementation for ESP8266 too - like with UART/SPI etc. The user shouldn't need to care what device something is running on - it should just work.

mvturnho, 2019-07-15:
I have been working on this thing and after checking the remote-transmitter/receiver approach I came up with te folowing.

For the protocol I sugest the standard canbus frames that basicaly consist of an id and a data part. I sugest just using the broadcast approach where the sender puts its frame on the bus and every interested receiver processes the data when it recognizes the canbus id. I think of adding the sender id when we use the 29 bit identifier setup.

For the canbus in esphome you can define several busses by using the platform option.
the platforms implement the hardware part.
```yaml
canbus:
  #SPI device 
  - platform: mcp2515
    id: first_canbus
    can_id: 600
    cs_pin: 14
    dc_pin: 27
    bus_speed: 500k
  #ESP32 hardware canbus
  - platform: esp32_can
    id: second_canbus
    can_id: 601
    rx_pin: 4
    tx_pin: 3
    bus_speed: 500k
```

Then we add `sensors` and `binary_sensors` to the specific canbus for the id's in the received frames.
The data part is used to set the sensor state. 
for binary_sensor:
0 = off
1 = on
2 = on -> off = (toggle)
for sensor we set the state with the received data.
```yaml
binary_sensor:
  - platform: canbus
    canbus_id: first_canbus
    can_id: 200
    id: key0
    name: "key0"
  - platform: canbus
    canbus_id: second_canbus
    can_id: 201
    id: key1
    name: "key1"
    on_click:
      then:
        - switch.toggle: switch1
```

Sending over the canbus is done by using the actions from a trigger in a sensor or `binary_sensor`.
So when `button` is clicked we send a frame with identifier 400 and the data that represents a click = 2  on the `first_canbus`. It seems that the `canbus.send` Action as shown below is not how esphome works with Actions. The default only allows for a components id. 
```yaml
binary_sensor:
  - platform: gpio
    id: button
    pin: 0
    on_click:
      then:
        - canbus.send: 
          canbus_id: first_canbus
          can_id: 400
```

So in components we need
- `canbus`
  - `__init__.py`
  - `binary_sensor.py`
  - `canbus.h` base class `Canbus`
  - `canbus.cpp`
  - `binary_sensor.h`
- `mcp2515`
  - `canbus.py`
  - `mcp2515.h`  `MCP2515 `subclass of `Canbus`
  - `mcp2515.cpp`
- `esp32_can`
  - `canbus.py`
  - `esp32_can.h` `ESP32Can` subclass of `Canbus`
  - `esp32_can.cpp`

Now for the automation part, the actions are all in the `automation.h`. Should the `canbus::SendAction<>` be added to this? And then how will the` canbus->send(id,data)` be called from the action?

in [my canbus branch](https://github.com/mvturnho/esphome/tree/canbus) I have something working for the python part that just generates the `main.cpp` with the folowing yaml setup:
```yaml
esphome:
  name: canbus
  platform: ESP32
  board: nodemcu-32s

spi: !include includes/esp32_spi.yaml

# Enable logging
logger:
  level: DEBUG

canbus:
  - platform: mcp2515
    id: first_canbus
    can_id: 600
    cs_pin: 14
    dc_pin: 27
  - platform: esp32_can
    id: second_canbus
    can_id: 601
    rx_pin: 4
    tx_pin: 3

binary_sensor:
  - platform: canbus
    canbus_id: first_canbus
    can_id: 200
    id: key0
    name: "key0"
  - platform: canbus
    canbus_id: second_canbus
    can_id: 201
    id: key1
    name: "key1"
  - platform: gpio
    id: button
    pin: 0
    on_click:
      then:
        - canbus.send: button
        - switch.toggle: switch1

switch:
  - platform: gpio
    pin: 1
    id: switch1
```

mvturnho, 2019-07-16:
@OttoWinter I am running in some problem with the python code. When I add an mqtt configuration to the yamle I get:

```
WARNING Components 'mqtt_mqttbinarysensorcomponent' were never registered. Please create a bug report
WARNING with your configuration.
```

I have been trying to solve this but without result. I made my canbus a platform component, with mcp2515 and esp32_can as my platforms. Both have binarysensors for a start.

the code for [canbus is here](https://github.com/mvturnho/esphome/tree/canbus):

rlust, 2019-08-01:
Please let me know when you have code to test. Vey interested in CAN bus integration!

mvturnho, 2019-08-01:
I have the transmitter part ready and started on the receiver part. I started with the mcp2515 CAN device. Progress is slow for I have limited time.

Checkout the canbus branch in my esphome fork for the code.

It sends data over the CAN bus from sensor automation. For exeample, when you add the on_click you can send the sensor state with a CAN message.

canbus:
  - platform: mcp2515
    id: first_canbus
    sender_id: 0x0A
    cs_pin: 15
    bit_rate: 125KBPS

binary_sensor:
  - platform: gpio
    id: button
    name: button
    pin: 
      number: 0
      inverted: True
    on_press:
      then:
        - canbus.send:
            canbus_id: first_canbus
            can_id: 401
            data: !lambda
              return {0x15, (uint8_t) id(button).state};
    on_release:
      then:
        - canbus.send:
            canbus_id: first_canbus
            can_id: 402
            data: !lambda
              return {0x016, (uint8_t) id(button).state};
    on_click:
      then:
        - canbus.send:
            canbus_id: first_canbus
            can_id: 403
            data: [0x00,0x00,0x25,0x26,0x27,0x28]

mvturnho, 2019-08-15:
Ok the canbus works now for sending en receiving. All sending en receiving is done from the automations from esphome. So when a button is pressed you can send its (or other sensor) state over canbus. 
And for the canbus you can register id's, so when this id is received you can execute some action on lights or switches etc.

I am looking forward to your feedback and new ideas.

You can clone and test from my forked repo. Use branch canbus.
from gitbash;

```bash
git clone https://github.com/mvturnho/esphome.git
cd esphome
git checkout canbus
pip install -e .
```

As to be suspected you neet two esp devices with mcp2515 Canbus controllers. When you order the cheap mcp2515 devices you neet to be aware that these are 5v boards. To use them with 3.3V esp you need to do some adjustments to the board.

use these pins for your spi on esp8266

```yaml
  clk_pin: D5 #gpio14
  mosi_pin: D7 #gpio13
  miso_pin: D6 #gpio12
```

The folowing yaml I used for the first device:

```yaml
esphome:
  name: canbus
  platform: ESP8266
  board: nodemcuv2

wifi: !include includes/wifi_iot.yaml
mqtt: !include includes/mqtt_iot.yaml
ota:  !include includes/ota.yaml
spi: !include includes/esp8266_spi.yaml

# Enable logging
logger:
 level: DEBUG

light:
  - platform: monochromatic
    name: "Lights"
    id: light_1
    output: output_component1

# Example output entry
output:
  - platform: esp8266_pwm
    id: output_component1
    pin: 
      number: 16
      inverted: true

canbus:
  - platform: mcp2515
    id: first_canbus
    sender_id: 10
    cs_pin: 15
    bit_rate: 125KBPS
    clock: 8MHZ
    mode: NORMAL
    on_frame:
      - can_id: 500
        then:
          - lambda: |-
              std::string b(x.begin(), x.end());
              ESP_LOGD("canid 500", "%s", &b[0] );
          - light.turn_off: light_1
      - can_id: 501
        then:
          - light.turn_on:
              id: light_1
              brightness: !lambda "return (float) x[0]/255;"
      - can_id: 502
        then:
          - light.turn_on:
              id: light_1
              brightness: !lambda "return (float) x[0]/255;"

binary_sensor:
  - platform: gpio
    id: button
    name: button
    pin: 
      number: 0
      inverted: True
    on_press:
      then:
        - canbus.send:
            canbus_id: first_canbus
            can_id: 401
            data: !lambda
              return {100, id(button).state};
    on_release:
      then:
        - canbus.send:
            canbus_id: first_canbus
            can_id: 402
            data: !lambda
              return {255, id(button).state};
    on_click:
      then:
        - canbus.send:
            canbus_id: first_canbus
            can_id: 400
            data: "sender  "
```

this yaml I used on my second device:

```yaml
esphome:
  name: canbus_receive
  platform: ESP8266
  board: nodemcuv2

wifi: !include includes/wifi_iot.yaml
mqtt: !include includes/mqtt_iot.yaml
ota:  !include includes/ota.yaml
spi: !include includes/esp8266_spi.yaml

# Enable logging
logger:
  level: DEBUG

light:
  - platform: monochromatic
    name: "Lights"
    id: light_1
    output: output_component1

# Example output entry
output:
  - platform: esp8266_pwm
    id: output_component1
    pin: 
      number: 16
      inverted: true

canbus:
  - platform: mcp2515
    id: first_canbus
    sender_id: 0x0A
    cs_pin: 15
    bit_rate: 125KBPS
    clock: 8MHZ
    on_frame:
      - can_id: 400
        then:
          - lambda: |-
              std::string b(x.begin(), x.end());
              ESP_LOGD("canid 400", "%s", &b[0] );
          - light.turn_off: light_1
      - can_id: 401
        then:
          - light.turn_on:
              id: light_1
              brightness: !lambda "return (float) x[0]/255;"
      - can_id: 402
        then:
          - light.turn_on:
              id: light_1
              brightness: !lambda "return (float) x[0]/255;"

binary_sensor:
  - platform: gpio
    id: button
    name: button
    pin: 
      number: 0
      inverted: True
    on_press:
      then:
        - canbus.send:
            canbus_id: first_canbus
            can_id: 501
            data: !lambda
              return {100, id(button).state};
    on_release:
      then:
        - canbus.send:
            canbus_id: first_canbus
            can_id: 502
            data: !lambda
              return {255, id(button).state};
    on_click:
      then:
        - canbus.send:
            canbus_id: first_canbus
            can_id: 500
            data: "michiel "
```

speters, 2020-10-30:
The aforementioned "canbus" branch at https://github.com/mvturnho/esphome.git does not exist.
Is the code still available anywhere? It came to my mind when thinking of a generic NMEA2000 sensor device...

fehli, 2020-11-01:
Thank you for the good work.
I was able to implement it. But now I've had a problem for days. I am trying to log the canbus of my heater. 
[03:54:54] [D] [canbus: 047]: received can message can_id = 203 size = 1 data = 59

So far so good, but how do I get the data bytes within my yaml to send it to an entity?

Thx

pergolafabio, 2020-11-02:
hi, i am verry interested in this
i have at home an NodeMcu ESP8266 and a MCP2516
my domotica system is based on canbus, i can already use it with an API (not canbus) to turn on lights/powersockets
The only thing i am missing is to intercept the presses on the wall switches ... if i could intercept them, i can use the wall switches for something else...

so i want to hookup and esp2866 and a mcp2516  , can someone point me to the wiring diagram?
but more important, if i am hooked up wit the wires, i dont know yet what i am expecing to receive when i press on a switch
i need to log first and see what happens when i press on wall switches... how can i found out that information? 

thnx for the help and advice!

pergolafabio, 2020-11-09:
@fehli  @mvturnho 

can you quide me? i am interested in this project, basicly my domotica system is based on canbus, so i only want to READ stuff
i have an esp8266 and mcp2515
can you show me a wire diagram?
also , how can i do a candump / log? i dont know yet the information from the canbus, how can i see the candump? so i can make sensors for it?

thnx!

aerodolphin, 2021-09-27:
Any developments on this? It's possible to use the internal can controller of the ESP32 and only an external can transceiver?

michaelansel, 2021-09-27:
It's definitely possible to use the built-in controller (an external transceiver is always required for the differential signaling). I put together a read-only implementation a while back for a project, but never got around to finishing it and cleaning it up for PR submission: https://gist.github.com/michaelansel/e3109a8513fa87587a9f70c8fa62f63b

cbialobos, 2021-11-24:
> @michaelansel nice to hear about that. Let's hope someone in a near future do the coding. Will be a major improvement to develop some hardware for the nodes on ESPHpome running on ESP32 devices.

I'm volunteering :) a part of the work is already done, inspired by @michaelansel

aerodolphin, 2021-11-24:
@cbialobos, very good news in the morning. Thank you for the help.

Have the hardware for the testing?

pergolafabio, 2021-11-24:
Just for info, HassOs has also driver support for USB devices like canable... I submitted that PR a year ago

cbialobos, 2021-11-24:
> 

I have a m5stack atom with the canbus unit (https://shop.m5stack.com/products/atom-canbus-kit-ca-is3050g).
On top of that, I have an rpi with a canbus hat so I can control the behavior of my development.

pergolafabio, 2021-11-24:
yeah i load a custom component/script upon start HA, i than load the drivers on HassOS with modprobe 
i use slcan drivers, native drivers dont need to be loaded ... i included several in HassOS

aerodolphin, 2022-01-06:
Regarding to canable hardware, it's open source?

What you have is from canable.io. I saw this about:

[https://github.com/makerbase-mks/CANable-MKS](https://github.com/makerbase-mks/CANable-MKS)

And this on Aliexpress:

[https://www.aliexpress.com/item/1005003746105255.html?spm=a2g0o.productlist.0.0.3b0c19b9VbutcT&algo_pvid=bb747a16-6841-4047-a544-b8ab86ef144f&algo_exp_id=bb747a16-6841-4047-a544-b8ab86ef144f-2&pdp_ext_f=%7B%22sku_id%22%3A%2212000027028724905%22%7D&pdp_pi=-1%3B29.26%3B-1%3BEUR+1.63%40salePrice%3BEUR%3Bsearch-mainSearch](https://www.aliexpress.com/item/1005003746105255.html?spm=a2g0o.productlist.0.0.3b0c19b9VbutcT&algo_pvid=bb747a16-6841-4047-a544-b8ab86ef144f&algo_exp_id=bb747a16-6841-4047-a544-b8ab86ef144f-2&pdp_ext_f=%7B%22sku_id%22%3A%2212000027028724905%22%7D&pdp_pi=-1%3B29.26%3B-1%3BEUR+1.63%40salePrice%3BEUR%3Bsearch-mainSearch)

All should work, right?

pergolafabio, 2022-01-06:
Yes , drivers are now loaded in hassos

aerodolphin, 2022-02-07:
> @aerodolphin: Did you get the MKS pro from Aliexpress to work with ESP32-devices?

Not yet. I'm trying to understand first if this is the best way for a Smart Home wired system. And if will work as needed. Fefore invest more money on the Hardware.

aerodolphin, 2022-02-07:
Any progress about CANBUS on ESPHome? Anyone have working examples to show and inform how It's working?

nagyrobi, 2022-07-07:
Implemented ESP32 integrated CAN controller in https://esphome.io/components/canbus.html#esp32-can-component

closed 2022-07-07 by nagyrobi