esphome/feature-requests #300: ina129

pauldogg, 2019-06-27:
**Describe the problem you have/What new integration you would like**

I would like to integrate into ESP home and retrieve readings from the ina169 chipset
**Please describe your use case for this integration and alternatives you've tried:**

Similar to the ina219, i would please like support for the ina169 which has more amperage, better suited for a solar panel
**Additional context**

There is already an adafruit library

pilotak, 2019-07-28:
If i'm not mistaken, INA169 is analog output IC so you can measure it with ADC

```yaml
sensor:
  - platform: adc
    pin: A0
```

OttoWinter, 2019-07-28:
@pilotak Is correct, this is an analog sensor, so you should use `adc`

closed 2019-07-28 by OttoWinter