esphome/feature-requests #301: Home Assistant events as Automation Triggers

Anonym-tsk, 2019-06-28:
**Describe the problem you have/What new integration you would like**
I want to use [Home Assistant events](https://www.home-assistant.io/docs/configuration/events/) as Automation Triggers in ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**
I can import `sensor` or `switch` from Home Assistant to ESPHome and subscribe to state changes, but i can't subscribe to other events, for example `telegram_callback`.

**Additional context**
Example configuration:
```
api:
  events:
    - event: telegram_callback
      then:
        - light.turn_on:
            id: my_light
            brightness: !lambda 'return event.data.brightness;'
```

OttoWinter, 2019-06-28:
Just create an automation in HA that subscribes to the necessary event and calls the ESPHome service with the data.

I will close this one because:

 1. Already possible with few lines of HA automation.
 2. And the syntax would never be as nice as API services because the schema (and data types) are not known. The syntax you showed in your example would never be possible in C++ for example.

closed 2019-06-28 by OttoWinter

Anonym-tsk, 2019-06-28:
Ok. Maybe i'll try to do it myself %)