esphome/feature-requests #302: Safe boot mode

Anonym-tsk, 2019-06-28:
**Describe the problem you have/What new integration you would like**
Detect problems in runtime and boot in safe mode with only `wifi` and `ota` components enabled.

**Please describe your use case for this integration and alternatives you've tried:**
I can write bad (but valid) yaml config, flash it over the air and my device will not boot.

**Additional context**
I have a few embedded device and after bad flashing i need to disassemble device and flash it with wire. I really need possibility to boot device in safe mode and flash it over the air.

OXOTH1K, 2019-06-28:
Great idea!

to4ko, 2019-06-28:
Want this one too!

OttoWinter, 2019-06-28:
Already exists and enabled by default. See ota safe mode.

closed 2019-06-28 by OttoWinter

glmnet, 2019-06-28:
What would be different from current safe mode? https://esphome.io/components/ota.html

Anonym-tsk, 2019-06-28:
@OttoWinter i'm not sure that `safe_mode` works.
Last 3 days i have one problem - I flash device over the air and... it's offline. In the evening i fix problem in config and flash it by wire. Right now one my device is offline. And yes, `safe_mode` is enabled.

OttoWinter, 2019-06-28:
@Anonym-tsk Then you might have an issue with WiFi - see the pinned issue in the bug tracker. If wifi doesn't work of course the device will also not be able to connect to the network for ota safe mode.

Anonym-tsk, 2019-06-28:
@OttoWinter no-no, it's not wifi problem. For example i can broke device with [Custom Sensor](https://esphome.io/components/sensor/custom.html) (it's easy), but i can do it with embedded components too.

OttoWinter, 2019-06-28:
@Anonym-tsk Then you should create an issue with the code that causes it and the logs with it. The feature already exists, if there's a bug please open a bug report.