esphome/feature-requests #303: Default AP ssid

NoNaMoe, 2019-06-28:
**Describe the problem you have/What new integration you would like**

Would be neat/less cluttered if AP (Access Point) SSID defaults to the ESPHome node name, rather than being a required field.

Example config:
```
esphome:
  name: dryer_pow_monitor
  platform: ESP8266
  board: esp01_1m

wifi:
  ap:

...
```

**Please describe your use case for this integration and alternatives you've tried:**
Seems like it would be a sane default to have.

Alternative might be to use a template/lambda, but ive tried this. ap.ssid is currently not templatable (why would it be):
```
wifi: [source .\dryer_pow_monitor.yaml:16]
  ap:  [source .\dryer_pow_monitor.yaml:17]

    This option is not templatable!.
    ssid: !lambda |-
      return "blah"; [source .\dryer_pow_monitor.yaml:17]
```
... and there doesnt seem to be a easy way to get 'name' from the Core/esphome component API anyway?

closed 2019-06-28 by close-issue-app[bot]

NoNaMoe, 2019-06-28:
... and looks like I just found a bug in GitHub: No template was provided when logging in to start a new issue.

Also bot is unclear: Do I just hope mantainer sees this issue, or should I create a new, duplicate issue using the template?

OttoWinter, 2019-06-28:
Usually create a new issue.

In this case though: don't, i will close it again because the requested stuff is already possible: please see substitutions.