esphome/feature-requests #304: UART modes other than 8N1

jplitza, 2019-06-28:
**Describe the problem you have/What new integration you would like**

A parameter to the uart component that makes it possible to use other other modes than 8N1 (in particular 7E1) would be great.

**Please describe your use case for this integration and alternatives you've tried:**

My power meter has a D0 interface and speaks OBIS at 9600 baud with 7E1 (7 bits per character, even parity, one stop bit) over that interface.

**Additional context**

Should be relatively easy by using the parameter [here](https://github.com/esphome/esphome/blob/1ca241615d1243d0f58833caf8711c56a0e83795/esphome/components/uart/uart.cpp#L32) (and all subsequent calls to `->begin` for ESP8266), at least for hardware UART.

MisterScience5, 2019-07-12:
I'd love this as well, as I'm currently struggling to integrate my meter with M-Bus, which operates at 2400 8E1

RaveGun, 2019-08-27:
+1 to this.
I need to use the 7E1 for some IR communication with the power meter.
Also I need to be able to setup the TX and RX default pin levels.

OttoWinter, 2019-08-27:
> I need to use the 7E1 for some IR communication with the power meter.

IR communication over serial? For IR you would usually use remote_transmitter component (that can generate arbitrary signals)

RaveGun, 2019-08-27:
> 
> 
> > I need to use the 7E1 for some IR communication with the power meter.
> 
> IR communication over serial? For IR you would usually use remote_transmitter component (that can generate arbitrary signals)

Thank you for the reply.

Yes, it is a serial communication that is using the IR like this:
https://www.rudiswiki.de/wiki9/VolkszaehlerACE3000

I would like to use the HW UART to avoid the bit banging solution :)

OttoWinter, 2019-08-27:
> I would like to use the HW UART to avoid the bit banging solution :)

Is this on ESP32? On ESP32 remote_* is already done with the remote peripheral (so no bit-banging). On ESP8266 you would probably be forced to use bit-bang anyway (unless you use special pins).

jplitza, 2019-08-28:
> I need to use the 7E1 for some IR communication with the power meter.

This was my exact use case, but 7E1 fortunately also has 8 bits, which means that you can manually compute the parity:
```C
static inline bool parity(char v) {
    // source: http://www.graphics.stanford.edu/~seander/bithacks.html#ParityParallel
    v ^= v >> 4;
    v &= 0xf;
    return (0x6996 >> v) & 1;
}
```
Afterwards, you only have to bitwise-and with `0x7f` if the parity bit matches:
```C
if (parity(c)) {
    c &= 0x7f;
} else {
    ESP_LOGI("serial", "parity error");
}
```

nldroid, 2019-12-21:
+1 for this
(Older) Dutch powermeters use 7E1 as well on their serial P1 port. You suggested to use UART instead of EspSoftwareSerial but we can't if 7E1 is not supported..

cklit, 2019-12-30:
I would love to have 7N1 functionality :) Tried to send it using the remote transmitter, but I can't get it precise enough.

jmartens, 2020-10-21:
I believe this is possible since fb2b7ade41dc3f5fae8a68e034b6506bf5902b0b as added per #1024 using the following options
- `data_bits`, ranging from 5 to 8
-   ##`parity`, none, even or odd

jplitza, 2021-11-25:
> I believe this is possible since [fb2b7ade41dc3f5fae8a68e034b6506bf5902b0b](https://github.com/esphome/esphome/commit/fb2b7ade41dc3f5fae8a68e034b6506bf5902b0b) as added per #1024 using the following options
> 
>     * `data_bits`: ranging from 5 to 8
> 
>     * `parity`: none, even or odd

Correct.

closed 2021-11-25 by jplitza