esphome/feature-requests #305: Automation for camera

petrepa, 2019-06-30:
**Describe the problem you have/What new integration you would like**

Would like the possibility to turn off camera features and activate them via automations. A button press can for example activate the camera for a given intervall.

Something like this:
``` 
 - platform: gpio
    pin:
      number: GPIO34
      mode: INPUT_PULLUP
      inverted: True
    name: doorbell button
    on_press:
      then:
        - camera.turn_on: my_camera
        - delay: 30s
        - camera.turn_off: my_camera
```

**Please describe your use case for this integration and alternatives you've tried:**

I would use this for my ESP based video doorbell, so the camera is only active for some while after the doorbell button is pressed. This so I will not have surveillance out onto the public street continuously, which can be illegal some places.  

**Additional context**

OttoWinter, 2019-07-01:
Well, if nothing accesses the camera image it also won't capture any images (if you set idle framerate to 0)

Seems like this should be solved in the frontend side rather than with commands.

petrepa, 2019-07-01:
Uuuh! Thats smart!

closed 2019-07-01 by petrepa