esphome/feature-requests #307: The KNX IP Protocol

dcec, 2019-07-01:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**
The KNX IP Protocol is an international open standard for smart homes and smart buildings automation. It is a decentralized system.
Several home automation systems have KNX support. 

**Additional context**

brandond, 2019-07-01:
This feels like something that would best be integrated directly into Home Assistant. How would you see this working with esphome?

dcec, 2019-07-02:
KNX is one of the most common protocols for commercial and domestic building automation.
Is also possible to integrate on existing wired knx using a coupler like this project:
http://www.intranet-of-things.com/smarthome/infrastructure/knx/arduino/

OttoWinter, 2019-07-03:
@dcec Yes, that might be true - but please also explain why this would specifically be required in ESPHome directly as @brandond asked.

With these things it's probably a lot more useful to do all KNX stuff in Home Assistant (as that will always be more powerful and feature-rich than ESPHome could ever implement KNX). ESPHome is not supposed to be used on its own - it works best with HA by its side.

dcec, 2019-07-05:
for example using esp32 for send xiaomi_mijia sensor data on knx network without Home Assistant

ricreis394, 2020-03-09:
@OttoWinter With knx you can communicate between esphome nodes
Here you can see more about knx -> [https://github.com/arendst/Tasmota/wiki/KNX-features](https://github.com/arendst/Tasmota/wiki/KNX-features)
An example use case: Control a light from node X, with a button in node Y

webfrank, 2020-06-20:
Hi I use knx on tasmota without any broker and would be useful to do the same on esphome. I can deploy a totally autonomuos automation

DocDoo, 2021-07-05:
I would like this feature as well as it, like already mentioned, will allow me to improve the robustness of my setup by having ESPHome nodes interact directly with KNX (e.g. to provide the other KNX components with sensor information) as a first step for automation in a new build I'm planning.
It will then be easy for me, in a second step, to add Home Assistant for advanced automation and integration scenarios in a future improvement to my installation.

RichieRich69, 2021-09-29:
This is incredibly useful as a fallback when home assistant is unavailable ea. When there is a update or home assistant falls to start for some reason.

DocDoo, 2022-07-05:
> I would argue that this issue is not closed since the external component only implements KNX TP but the issue is about KNX IP protocol. KNX IP does not require additional hardware (bus coupler) as it communicates directly over IP protocol (WiFi, Ethernet).

I agree with this argument, but very appreciate the excellent work done by @fxmike08 👍👍

kmplngj, 2022-08-17:
I found this issues because I would like to use the new _Feedback Cover_ with sensors for Open/Closed directly on the esp and KNX IP for KNX relays to switch open/closes actions on the door motor.

borispulyer, 2023-02-05:
@fxmike08 I really appreciate your component. Is there any chance to get some more configuration examples, eg. for sending telegrams?

fxmike08, 2023-02-06:
Hi Boris,

You have below an example on how to send knx telegrams. For some reason you have to add **id** to **knx** component in order to access it.

Please have in mind that if ESP start to crash and restart when sending telegrams, it means that you don't have proper connection between serial and tp-uart. 

```
  external_components:
    - source:
        type: git
        url: https://github.com/fxmike08/esphome
        ref: KnxComponent
      components: [ knx ]
...
  uart:
    - id: uart_bus
      tx_pin: 1
      rx_pin: 3
      baud_rate: 19200
      parity: even
      debug:
        direction: BOTH
        dummy_receiver: false
        after:
          delimiter: "\n"
        sequence:
          - lambda: UARTDebug::log_binary(direction, bytes, 32);
...
  knx:
    id: knxd
    uart_id: uart_bus
    use_address: 10.10.1
    listen_group_address: ["0/0/3", "0/0/1"]
    lambda: |-
      KnxTelegram* telegram = knx.get_received_telegram();
      ESP_LOGD("KNX", "group: %s",telegram->get_target_group());
      if(telegram->get_target_group() == "0/0/3" ){
        if(telegram->get_bool()){
           ESP_LOGD("KNX", "KNX ON");
        } else {
           ESP_LOGD("KNX", "KNX OFF");
        }
          
      }
        
      return;
...
  api:
    services:
      - service: knx_write_group_bool
        variables:
          address: string
          value: bool
        then: 
          - lambda: |-
              id(knxd).group_write_bool(address.c_str(), value);
...
  - platform: gpio
    name: $friendly_devicename Relay 2
    id: relay_2
    pin:
      number: 19
    on_turn_on:
      - lambda: |-
          id(knxd).group_write_bool("0/0/3", true);
```

borispulyer, 2023-02-06:
Thank you very much @fxmike08 - this helps me a lot. Great work!

platini76, 2023-03-13:
how to install knx component into esphome using homeassistant interface?

fdepalo, 2023-06-07:
@fxmike08 thank you for your effort, can we use a siemens bcu 5wg-1117-a2b12 directly to esp32 serial or it is just for tpuart chip like nanobcu ?

fdepalo, 2023-06-07:
mistery solved, i am able to transmit but not to receive without the level shifter, thank you ;)

fdepalo, 2023-06-10:
> Hi Boris,
>
>
>
> You have below an example on how to send knx telegrams. For some reason you have to add **id** to **knx** component in order to access it.
>
>
>
> Please have in mind that if ESP start to [...]

@fxmike08 thank you again, is there something i can do to help you with debugging/development? The crash if tpuart is not connected is the only problem i am experiencing:)

fxmike08, 2023-06-23:
@fdepalo Thank you for willingness to help, but at this point I don't know if you can help with something.  When I'll have time I'll try to find a solution for serial crash (There is no way to check if serial is working).

PaoloTK, 2023-07-31:
> @fdepalo Thank you for willingness to help, but at this point I don't know if you can help with something. When I'll have time I'll try to find a solution for serial crash (There is no way to check if serial is working).

Hi @fxmike08, I'm working on a similar KNX integration for WLED. Can you explain where you sourced the library you're using and would it be possible to upload it to GitHub separately? I've found an older version of the same library hosted here:

https://github.com/majuss/KnxTpUart

fxmike08, 2023-08-14:
@PaoloTK please find [here ](https://github.com/fxmike08/knx) the requested info. 

Yes, [KnxTpUart ](https://github.com/majuss/KnxTpUart) is the library used and modified to compile and for respecting the coding style of the esphome. Let me know if you need more info. 

Regards.

nerdoc, 2023-10-03:
A really good reason to establish this component is, if (like in my house) an esphome component is not reachable all the time using WiFi (e.g. to far away), or (like here) Wifi is not always switched on. I could use a WT32-ETH01 with a LAN connection, - but: I have KNX cables literally everywhere in my house, and LAN cables only where I need them.

So using the KNX bus as wired connection to Homeassistant would be the perfect match. I just need to send updates to the server. And everytime I switch off WiFi, the continual capturing of sensor data is lost.

So I would definitely second a knx esphome component.

fxmike08, 2023-11-25:
@nerdoc Haven't test with other bus couplers, but I suspect it would work.

nerdoc, 2024-05-20:
Any updates on this? Again, esphome would fully replace things like [KONNECTING](https://www.konnekting.de/konnekting-lernen/l1-knx-mit-arduino/) or [OpenKNX](https://github.com/OpenKNX) where the whole ETS functionality must be rebuilt from scratch. Esphome has everything it needs, the compiler, the configuration utilities, components, a yml language to define code.

It would be an ideal connection between these both worlds.

* Install software updates using Homeassistant/esphome when WIFI is available.
* The arduino device itself gets current from the KNX bus, no 5V charger etc needed.
* access to the KNX bus done by Siemens bus coupler.
* Messages are sent as KNX telegrams, so WIFI is not needed  all the time. 
  * other KNX devices could respond directly
* Home Assistant can react on KNX messages nearly "natively" because KNX is perfectly integrated into Home Assistant
* Fast config/code updates are possible when turning on WIFI
So these hybrid esphome devices would be easy and flexible to program (compared to arduinoIDE with native KNX libraries) and maintain (OTA updates)

nerdoc, 2024-05-20:
There is also a (German) discussion [for OpenKNX stack and esphome](https://knx-user-forum.de/forum/projektforen/openknx/1828906-openknx-und-esp-home)

fxmike08, 2024-05-20:
Hi @nerdoc ,

The "Esphome - knx" component is available on [github ](https://github.com/fxmike08/custom-esphome/tree/main/components/knx). It can be used as external component. 

> The arduino device itself gets current from the KNX bus, no 5V charger etc needed.

Depends on your scenario. "Each KNX device consumes 10 mA from the power supply for transmitting and receiving a signal on the bus".  
If you will try to power an ESP32/ESP8266 from bus coupler it will not work, the power composition of ESP32 is much higher (peak 250 mA).
If you connect directly to KNX TP bus, yeah sure it will work, but will require some electronic PCB development for that.

> Home Assistant can react on KNX messages nearly "natively" because KNX is perfectly integrated into Home Assistant

You need a "knx ip interface" for making Home Assistant capable to communicate with KNX "natively". 

> would it be possible to create a HACS module?

Why would you need a HACS module? You add the knx component in the esphome yaml file, and that's it. I don't have to much experience with HASC, and I don't see the benefit of that.

nerdoc, 2024-05-21:
> Hi @nerdoc ,
>
> The "Esphome - knx" component is available on [github ](https://github.com/fxmike08/custom-esphome/tree/main/components/knx). It can be used as external component.
>
>
> Depends on your [...]

Yes, but for low-level applications (reading from sensors 1x/min, writing at bus) the power consumption of my e.g. "Arduino nano every" is low enough.

> > Home Assistant can react on KNX messages nearly "natively" because KNX is perfectly integrated into Home Assistant
> 
> You need a "knx ip interface" for making Home Assistant capable to communicate with KNX "natively".

Sure, with "native" I meant the knx IP interface. But it's included/can easily be added via GUI.

> > would it be possible to create a HACS module?
> 
> Why would you need a HACS module? You add the knx component in the esphome yaml file, and that's it. I don't have to much experience with HASC, and I don't see the benefit of that.

Oh, then I need to learn here a bit, sorry. I didn't know you can add external components. I'll have a look into that.

henfri, 2024-07-27:
`Yes, but for low-level applications (reading from sensors 1x/min, writing at bus) the power consumption of my e.g. "Arduino nano every" is low enough.`

But that ist Not **esp**Home then.
There are projects to communicate between KNX and arduino though.