esphome/feature-requests #309: Support for Multiple Power Supplies

poldim, 2019-07-03:
**Describe the problem you have/What new integration you would like**
powersupply function on outputs does not allow a list of power supplies

**Please describe your use case for this integration and alternatives you've tried:**
I have multiple power supplies, some for 12v, some 24, and some that need to be turned on together to create 48 (2x24).  

**Additional context**
This is going onto my whole house power controller so I'd like to be able to have various power supplies.  In the future, I'd like to build the hardware/software in such a way that it can measure the load and only bring up the required number of power supplies.

OttoWinter, 2019-07-05:
Well to me the solution should definitely not be to accept multiple power supplies in all `power_supply:` options, but to make `PowerSupply` an abstract class where the user can customize the behavior with custom code.

Other (probably simpler) option would be for you to just use some external circuitry for that (just a few transistors will do, basically just an OR gate).

poldim, 2019-07-08:
I also just tried using a template switch that is setup for both pins however I get the error that the `powersupply` component needs a pin an not a switch.  Is there any easy workaround for it to accept a switch instead of a pin?

`    Cannot resolve pin name 'switch.relay1and2' for board d1_mini.`

glmnet, 2019-07-08:
I had this problem too. May be power supply should support an output instead of a pin 
I solved adding an automation in esphome: loop:

OttoWinter, 2019-07-10:
@glmnet Yes, I know (also remember considering that when I first implemented power supply ages ago :) - problem however is that users could try to add the power supply to the power supply *output* component, so infinite recursion. That would not be simple to catch in a clean way. I guess one way would be to just expect users to not do that :)

For sure allowing any output would be the cleanest solution.

poldim, 2019-07-14:
I was thinking of using conditions that would check if any of the lights are on and only then turn off the power supplies, but it seems like #98 is stopping me from doing this easily.

glmnet, 2019-07-14:
Well, I have this code working, it's been written for some old version of esphome, you can now take advantage of the light is on / off conditions. (see below)

````yaml
esphome:
  name: eh_planta_alta
  platform: ESP8266
  board: nodemcu
  on_loop:
    then:
      # set the common neutral wire relay for patio lights
      lambda: |-
        bool binary1;
        bool binary2;
        id(luz_patio_oeste).current_values_as_binary(&binary1);
        id(luz_patio_este).current_values_as_binary(&binary2);
        if (binary1 || binary2)
          id(gpio_relay_5).turn_on();
        else
          id(gpio_relay_5).turn_off();
```` 

This is why i like most of ESPHome you can always inject some C++ code (or a lot of it) and do whatever you want, or you can go with automatons and do lots of things too.

The following code is not tested, writing it here just for you:

````yaml:
esphome:
  on_loop:
    then:
      if:
        condition:
          light.is_on: my_light
      then:
        switch.turn_on: my_ps
      else:
        switch.turn_off: my_ps
```` 

You can experiment with more complex conditions using and / or

closed 2022-07-01 by nagyrobi