esphome/feature-requests #310: Suppor for I2C Based motor controller

res2cpu, 2019-07-04:
**Describe the problem you have/What new integration you would like**
I would like some kind of support for some kind of I2C based motor driver like the WEMOS D1 MINI MOTOR DRIVE SHIELD.

**Please describe your use case for this integration and alternatives you've tried:**
I have used this motor driver for roller blind controllers and as they use the I2C bus it saves pins, is small and cheap.  In my use case i control 4 motors, with a traditional motor driver i would need a minimum of 8 pins, this way i only require 2.

**Additional context**
A arduino library for this already exists [here](https://github.com/wemos/LOLIN_I2C_MOTOR_Library), but I've had no luck getting it to work with esphome (lack of knowledge on my part i believe)

kwdaily, 2020-08-09:
I've been working on tackling this myself. I have the driver set up as a custom component for now as well as ESP Rotary library for encoder feedback, using custom home assistant services to start, stop, and reverse the motor. But so far I cannot get the motor driver to respond. Will post update when I have a solution working, hopefully in the next couple of days.

noxhirsch, 2020-09-22:
@kwdaily Any news on this matter?

Th3H0ff, 2021-06-17:
*sigh* Are you telling me this issue/request is two years old? I just posted a request myself basically requesting the same thing. Damn

Zhuaiz, 2022-06-20:
same problem

kwdaily, 2022-06-25:
No, I ended up writing my own driver using the I2C commands from the
controller datasheet

inspectorgrow, 2023-03-14:
> No, I ended up writing my own driver using the I2C commands from the controller datasheet
> […](#)
> -- Kenneth Daily ***@***.***

Dont suppose you mind sharing it, do you?

kwdaily, 2023-03-18:
[sliding_glass_door_lock_YAML_edited.txt](https://github.com/esphome/feature-requests/files/11009453/sliding_glass_door_lock_YAML_edited.txt)
[sliding_glass_door_lock.h.txt](https://github.com/esphome/feature-requests/files/11009459/sliding_glass_door_lock.h.txt)

YAML is the config (minus my credential info), sliding_glass_door_lock.h.txt just needs the .txt removed to turn it back into a .h file, which is included with the YAML. Fair warning, I scrapped this approach for my application but it was working at the time. If you have any questions, I will do my best to answer them! Hope this helps!

inspectorgrow, 2023-03-18:
I managed to get them working by directly communicating with the 16-channel PCA9685 that sends PWM to the 2 x TB6612 on the board. Sending 0% or 100% PWM turns the channels off/on for the in1 and in2 for forwards and reverse and then variable PWM for the speed control pin on the TB6612s.