esphome/feature-requests #311: extend automation functions for controling lights

mvturnho, 2019-07-04:
**Describe the problem you have/What new integration you would like**

In the automations we have functions to:
- toggle lights
- switch light ON
- switch light OFF

I have a need to also set the folowing:
- brightness
- effect

I know you can do this using mqtt, but I also want to be able to set this based on a switch or sensor value. I gues we should then use the automation.

**Please describe your use case for this integration and alternatives you've tried:**

press a button and have an indicator led start blinking (using the strobe effect on a light)

**Additional context**

glmnet, 2019-07-04:
Are you aware you can do all this using lambdas?

mvturnho, 2019-07-05:
@glmnet  Yes I know, and I have tried to make one. But I was unable to determine how to set the brightness or effects from a lambda.
Could you help me out on this? All I need is same hint on what to call after I get the refference.

I tried the obvious ;-)
```
id(light0).set_brightness(255)    //DOES NOT WORK
id(light0).set_effect("strobe")   //DOES NOT WORK
```

OttoWinter, 2019-07-05:
Please see parameters to light.turn_on (and light.control) action.

closed 2019-07-05 by OttoWinter

mvturnho, 2019-07-06:
Got it, thanks, I should read the docs ;-)
Also this solves the blink when I set the strobe effect
    
```
  on_click:
      then:
        light.turn_on: 
          id: light5
          brightness: 50%
          effect: strobe
```