esphome/feature-requests #312: Create WifiAP if connection fails 4 times

marrobHD, 2019-07-04:
**Describe the problem you have/What new integration you would like**

A function that opens a wifi AP after 4 Wifi failed connection attempts, where you can enter your wifi ssid and password

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

A function that opens a wifi AP after 4 WLAN failed connection attempts, where you can enter your wifi ssid and password.
In this AP, if you have configured a web interface password, you have to enter this, and then come to an interface in which you can enter his Wifi SSID and Wifi password. Thereafter, the device should be restarted with this data.
In addition, after the 4 failed connection attempts should be a timer that is set to 3 minutes, and after the expiration of the 3 minutes restart.

OttoWinter, 2019-07-05:
Duplicate of https://github.com/esphome/feature-requests/issues/86

closed 2019-07-05 by OttoWinter

marrobHD, 2019-07-05:
In tasmota the command "WifiConfig 2" makes the same for esp32 and esp8266. With a webui