esphome/feature-requests #314: Support Add on / 3rd Party Components

timpur, 2019-07-07:
**Describe the problem you have/What new integration you would like**

Im trying to create some components for my own projects that would most likely not be valid for use officially ... thus it id like to have "custom components" that can be sourced from pip or github ...

**Please describe your use case for this integration and alternatives you've tried:**

The solution ive come up with involved two additions to the esphome config.
1. Allow custom namespaces to be used to search for components in addition to the existing two `"custom_components", "esphome.components"`
2. Allow pip modules to be installed as part of the project via config.

With these 2 features i can get pip to download my component module from git or pip and tell esphome to search for components under that custom module by adding the namespace.

**Additional context**

This is an initial implementation and would need some work to perfect, welcoming feedback :), thanks in advanced.

glmnet, 2019-07-07:
I like this. If the code is pulled from gh it would be nice to set the tag of the commit

timpur, 2019-07-08:
Since were just using pip, you can do most of these https://pip.pypa.io/en/stable/reference/pip_install/#examples

Eg install from github :)

OttoWinter, 2019-07-10:
Yes, I had something like that in mind too - however I decided not to do it for a couple of reasons:

 1. First, the ESPHome internal API needs to be stable - currently only the user-facing API can be considered mostly stable (and breaking changes are documented).
 2. Connected to (1): As with HA, it's a lot better to have all code in-tree, it's easy to change internal APIs and easy to analyze/change all parts of code at once.
 3. Ideally, the mechanism with which to install plugins would be GUI-based in dashboard (kinda like octoprint). Doing so via config is fine too but of course not as nice.
 4. Problem with GH is versioning: There would need to be format defined to allow selection of a specific version of a custom component
 5. All custom components should be required to use pinned versions. I don't want stuff to break for users if a custom component version change is pushed and code changes in background without user knowing.
 6. Install of these things is not that easy either - where to download GH data? what pip mechanism to use (entrypoints)?

All in all, lots of things to consider and to do before this is done.

Until then, I would just suggest using `custom_components` folder - the HA community has been using that already for a long time and while it's not perfect, it works well.

timpur, 2019-07-12:
@OttoWinter, where can i post, tanking everyone involved in the revamp of esphome. The new way of writing the python scripts and c++ code together is so nice !!! Hat off to you guys !! Hope to contribute back some day.

nagyrobi, 2022-07-01:
Implemented: https://esphome.io/components/external_components.html

closed 2022-07-01 by nagyrobi