esphome/feature-requests #315: ESPiLight Support - 433 MHz devices

denxhun, 2019-07-09:
**Describe the problem you have/What new integration you would like**
https://github.com/puuu/ESPiLight

ESPiLight is a great library support many 433.92 MHz devices. Not only input but ouput as well. (It is not my work, contributed only a very little piece.)

If EspHome would include that library it would be wonderful.

**Please describe your use case for this integration and alternatives you've tried:**
I have a few 433.92 MHz based temperature sensors and power switches. This library handles them well but if a powerful ESP32 device can handle them together with DHT and BLE based other sensors that would be great.

**Additional context**
I'm happy to contribute on the coding side and the testing side as well.

rradar, 2019-07-09:
Somehow related: #258

denxhun, 2019-07-10:
Yes it is similar but my suggestion is more exact. The mentioned ESPiLight package is able to handle 433 MHz based power outlets too. In my fork I started to work on it.

glmnet, 2019-07-10:
I'm still learning 433mhz stuff, little experience yet.
I can tell you ESPHome relies little on third party libraries, also there is already support for 433mhz stuff via the rc switch protocols implementation, i.e. the protocols are implemented in ESPHome code. Both receiver and transmitter are implemented (input and output)
What are the differences between already implemented rc switch in ESPHome and the ESPiLight support?

denxhun, 2019-07-10:
Thanks for your hint. I found the rc-switch implementation in ESPHome. 
I have a little experience in 433 MHz area (but definitely more than with ESPHome I have) and as far as I know RC-Switch and ESPiLight are toatally different solutions. While RC-Switch is used for power switches then ESPiLight has a wider range of encoder and decoder codes to support weather stations and other RF devices too beside the power swithces. Feel free to look at https://manual.pilight.org/protocols/433.92/index.html page.

OttoWinter, 2019-07-10:
I understand Pilight is a collection of many different protocols - of course all of these can be added into ESPHome, but a generic feature request like this one is not really helpful.

I'd rather have a feature request per protocol that would want to be supported. Then progress can be better tracked.

As @glmnet said, esphome doesn't use other libraries for this stuff. For remote_transmitter/receiver this is because ESP32 has a specialised remote peripheral, which most libraries do not support. So ESPHome has its own remote base that allows for constructing remote data independent of the backend.

closed 2019-07-10 by OttoWinter