esphome/feature-requests #316: Will Esphome support Notify like Telegram?

theniz, 2019-07-09:
**Describe the problem you have/What new integration you would like**

Now I was notified via telegram on Homeassistant when my node connected after boot.
But, what if Homeassistant was down? 
I think it would be best for Esphome to send some message directly to me about node's status. 
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

If Esphome could send some message, it could monitor HA's status.
Everyone doesn't always use Esphome with HA, so some notify module will be needed.

rradar, 2019-07-10:
Interesting. But I think the overhead for the small memory of the esp will be the limitating piece in this puzzle. Specially if it comes to encryption (I think telegram does not have it activated by default and if you want to use end-to-end encryption it's only possible in 1 on 1 chats).

But if it is somehow possible I would prefer a messenger/protocol which forces a end-to-end encryption like signal :+1:

OttoWinter, 2019-07-10:
I will close this.

I don't want esphome to be bloated with stuff that should be done through other systems (and other programming languages). Anything HTTP-related is one of those things, if you want to control stuff through HTTP you should use other systems (for example a second computer). ESPHome is fundamentally about connecting HW to systems.

But... btw you can use SMS in the next version via sim800l by @glmnet

closed 2019-07-10 by OttoWinter