esphome/feature-requests #318: Ai-Thinker Camera LED control

bwente, 2019-07-10:
**Describe the problem you have/What new integration you would like**

Support for the LED Flash/Lamp onboard the ESP32-CAM (AI-Thinker). I would like to be able to activate it manually or automatically in night time conditions.

**Please describe your use case for this integration and alternatives you've tried:**

It is GPIO 04

**Additional context**

rradar, 2019-07-10:
@bwente 

What happens if you map GPIO 04 as an output and add it to the light component? You should be able to activate it manually or add some rules to switch it automatically.

OttoWinter, 2019-07-10:
Correct, LED flash is already supported as you wrote yourself - just turn GPIO4 ON with a GPIO switch any time you want to have the flash enabled.

closed 2019-07-10 by OttoWinter

bwente, 2019-07-11:
Wow. Thanks. I didn't realize you could add multiple elements. I just started two days ago. ;)