esphome/feature-requests #319: Dmx output component for light / fan

TurtleForGaming, 2019-07-10:
**Describe the problem you have/What new integration you would like**
I have some max485 that I use to create dmx device for my light dmx is cool because its a bus system and require only 1 esp to control it I would really like an integration into esphome

**Please describe your use case for this integration and alternatives you've tried:**
Add a dmx platform for controlling for example fan speed or light intesity/color
may be a config could look like this :
`
light:
  - platform: dmx
    start_address: 0
    channel_pattern: RGBW
    name: "TestDMX"
    output: output_component1

fan:
  - platform: dmx
    start_address: 0
    channel_pattern: D
    name: "TestDMX"
    output: output_component1
`
(D for dimmer) And I didn't try to use anything else for dimming the fan since I don't really like messing with high voltage

**Additional context**

OttoWinter, 2019-07-10:
Related: https://github.com/esphome/esphome-core/issues/25

poldim, 2019-11-09:
@OttoWinter 
Just trying to understand if I should attempt to build this or if it would be a waste of my time:
ESP8266 + MAX485 to control a set of DMX lights
Create a custom component in ESPHome using the [DMXSimple library](https://github.com/PaulStoffregen/DmxSimple?files=1) as shown in [this video @ 10:20](https://youtu.be/4PjBBBQB2m4) 

Does anything hardware or architecture wise that would prevent me from getting this to work?

OttoWinter, 2019-11-12:
> Does anything hardware or architecture wise that would prevent me from getting this to work?

No, it should work.

The library you pointed to however does not seem to support ESP devices, you probably need to use another one.

cupertinomiranda, 2019-11-27:
Last couple of evenings I have been working on implementing support to DMX512 protocol.

[Here](https://github.com/cupertinomiranda/esphome/tree/dmx_support) is the code for further discussion and perhaps future merge request.
Implementation was inspired by modbus code. Since DMX512 differs from standard UART, this implementation requires some hardware hack which uses a port to ground the data signal for protocol start sequence.

Example YML:

```
uart:
  tx_pin: D0
  rx_pin: D1
  stop_bits: 2
  baud_rate: 250000
  id: uart1

dmx512:
  id: dmx
  start_tx_pin: D6
  uart_id: uart1
  # TODO: add a update_every X seconds just in case something resets.
  # TODO: Currently I am sending a packet only when an update happens.

output:
  - platform: dmx512
    channel: 1
    double_channel: False
    universe: dmx
    id: dmx_1

light:
  - platform: monochromatic
    name: "Test Lights"
    output: dmx_1
    id: light_test
    default_transition_length: 2s
```

asmaps, 2020-02-07:
Any news on this? I'd be interested in this feature as well. ESP8266 + MAX485 is exactly the setup I have. I'm using https://github.com/mtongnz/ESP8266_ArtNetNode_v2 at the moment, but it's not very reliable in my experience as I get frequent reboots of the esp with flickering of the lights included.

Tuckie, 2020-11-16:
Also interested in DMX output support!

andyboeh, 2021-11-06:
I'm also very much interested in DMX support and I had a look at the current solutions. All have their drawbacks:
  - @cupertinomiranda's solution requires an additional transistor to generate the necessary break signal. 
  - @jakosek's solution does not require additional hardware but is limited to the ESP8266
  - I have a working solution without additional hardware, but it is limited to the ESP32

For the software side, the additional transistor is by far the simplest solution. All other methods to generate the break signal (88us low on the Tx line) that I am aware of are more workarounds and would require some nasty changes to the UART component.

@OttoWinter I'd be interested in what you consider the simplest / cleanest solution, or in other words: How to best implement this?

jakosek, 2021-11-07:
@andyboeh I'm on ESP32 too, look at [dmx_no_uart](https://github.com/jakosek/esphome/tree/dmx_no_uart) branch. It uses most common ESPDMX library with no UART component as dependency. It should work both ESP32/8266. But still generates some stability issues... 

IMO brake signal isnt problem at all, but sending DMX frame on every loop.

andyboeh, 2021-11-07:
Ah, thanks for the hint, I didn't see your other branch! I'll have a look at it next weekend.

andyboeh, 2021-11-13:
For testing purposes, I created a custom component for DMX512. It uses the internal UART and generates the break signal by detaching and attaching the pin from/to the pin matrix. It is limited to the ESP32 using the Arduino framework (it should also work using IDF, but this is untested. Requires a change in the config validation if you would like to try it).

Code, README and example: https://github.com/andyboeh/esphome-dmx512

For now, I didn't want to change anything on the ESPHome code, thus the TX Pin needs to be configured twice and you need to figure out which hardware UART is used (it defaults to 1, which is usually the case if serial logging is enabled).

I only did limited testing using a Lite-Puter DMX dimmer pack, I can't say anything about the stability.

someweisguy, 2021-11-14:
@andyboeh this looks really awesome. I will find some time to play around with it!

I built a [DMX library](https://github.com/someweisguy/esp_dmx) for ESP32 that runs on ESP-IDF and Arduino. Some of the feedback I received included interest in porting it to ESPHome. The issue for me is that I am completely new to the ESPHome environment and I'm still wrapping my head around how to get a custom component running. My library doesn't depend on any existing UART code - I wrote a completely new UART driver for it that runs on the ESP32 hardware. So my library doesn't depend on any existing UART driver. Any chance that could be helpful here? It doesn't run on ESP8266, but maybe it would allow for DMX input/output on an ESPHome release branch?

andyboeh, 2021-11-14:
@someweisguy I'm not too familiar with ESPHome either, so what I'm doing might be wrong as well. However, my assumption is that every component that uses a hardware UART needs to go through ESPHome's UART implementation. Otherwise, the UART assignment is going to be messed up in case the user configures an additional UART.

That said, the main problem with using the ESPHome UART component is that it doesn't have support for sending the break signal. That's the reason for the awkward double-configuration of the TX pin in my implementation.

Your library supports a lot more than what I wanted to achieve - my goal is to be able to dim the living room lights :)

cristianm, 2021-11-18:
Also interested in DMX output support!

Tuckie, 2021-11-18:
For the record, this is what I'm hoping to be able to use:

* https://shop.m5stack.com/products/atom-tail485?variant=32169041559642
* https://shop.m5stack.com/products/atom-lite-esp32-development-kit

This makes for a very compact and simple setup that could be easily powered/embedded into a dmx fixture. The downside is I don't have any flexibility on the circuit. (no transistor)

jjk0giap, 2021-11-24:
@andyboeh 
Great work! On my setup I use 128 GPIO from i2c mcp expanders + about 40 dmx channels. On my code ESP32 sometime crash. After exception decode, problem is HardwareSerial.end() and begin(). 
With Your code so far no errors have occurred, but I modified max channel. With full 512ch frame, on my setup as above, there was API reconnections loop. Probably full frame takes too much time blocking main loop. 
Please consider adding setting for max channels, as DMX don't need full universe at once.

jakosek, 2021-11-24:
@andyboeh
Great work! On my setup I use 128 GPIO from i2c mcp expanders + about 40 dmx channels. On my code ESP32 sometime crash. After exception decode, problem is HardwareSerial.end() and begin().
With Your code so far no errors have occurred, but I modified max channel. With full 512ch frame, on my setup as above, there was API reconnections loop. Probably full frame takes too much time blocking main loop.
Please consider adding setting for max channels, as DMX don't need full universe at once.

andyboeh, 2021-11-24:
I have some kind of "throttling" functionality in the pipeline where it sends full frames only twice a second and not as fast as possible - I just haven't had the chance to test it (I'll push it afterwards). That would probably solve the problem as the `.flush()` calls are most of the time not needed. I'll try to make the "refresh rate" configurable.

Further, if I understood the protocol correctly, for new values it would be sufficient to only send the bytes up to the changed channel. That should improve the situation as well.

jakosek, 2021-11-24:
> only twice a second and not as fast as possible
no, no - it should be as fast as possible - You could try add some transition to light - higher fps is better.  
According to dmx timing, full 512 channel frame "takes approximately 23 ms to send, corresponding to a maximum refresh rate of about 44 Hz", but from esphome docs there is info that loop() shouldn't block longer than 20ms. In theory nothing happens if take some longer, but in practice I have API reconnections.

andyboeh, 2021-11-24:
> no, no - it should be as fast as possible - You could try add some transition to light - higher fps is better.

Maybe I wasn't clear enough about that: Changed values, i.e. transitions, are of course sent as fast as possible! But refreshing old / already set values can be slower.

andyboeh, 2021-12-04:
@jakosek I just pushed the promised update:
  - If values are unchanged, updates are sent roughly every 500ms
  - Changed values are pushed immediately (transitions etc.)
  - The message sizes is automatically adapted to the number of channels in use: If you only have one channel, the DMX messages are only 2 bytes long. That should improve performance!
  - Tested with a meanLED S1-DR two-channel DMX dimmer

jsimonetti, 2021-12-04:
@andyboeh 
Could you add an example board layout?
I have been having trouble getting this to work.

klaernie, 2021-12-04:
> Could you add an example board layout? I have been having trouble getting this to work.

If it helps you, that's what is running well on my end: https://github.com/klaernie/esphome-configs/blob/main/dmx01.yaml

andyboeh, 2021-12-04:
> Could you add an example board layout?

What exactly do you mean, a schematic diagram? Or the configuration file? An example configuration file is provided in the github repository.

poldim, 2021-12-04:
> > Could you add an example board layout?
> 
> What exactly do you mean, a schematic diagram? Or the configuration file? An example configuration file is provided in the github repository.

I’d love a schematic if you got one

andyboeh, 2021-12-04:
> I’d love a schematic if you got one

Please find attached the very simple schematic for the RS485 adapter that I'm using. A bit more background info: I use a WT32-ETH01 board that is plugged into a base board. On the base board is a 12-pin header where the RS485 adapter board is plugged into. The adapter board provides a RJ45 socket with the DMX signal. On the PCB, I also added a simple 3-pin screw terminal as an alternative.

[rs485_adapter.pdf](https://github.com/esphome/feature-requests/files/7654371/rs485_adapter.pdf)

jsimonetti, 2021-12-04:
I'm pretty sure it's my electric skills that are failing me now :(
I am trying to use one of these max485 boards: https://protosupplies.com/product/max485-ttl-to-rs-485-interface-module/
I hooked up the DI to the tx_pin and the DE to the enable_pin. I'm leaving the RE and RO floating.
Unfortunately nothing happens. GND is connected to RJ45 pins 7 and 8, and A is pin 1, B is pin 2 (according to ESTA standard). 
I hope someone here knows this particular max485 TTL board and can give some pointers on how to use it. I don't have a scope unfortunately, so cannot debug further.

I do appreciate your work with this module! Me failing is obviously a problem from my end.

klaernie, 2021-12-04:
@jsimonetti The things I'd try first are: check VCC on the max485 board. If it doesn't have enough voltage it can't put out enough voltage for the other side. Then, if you don't have a scope or logic analyzer you can try to swap a and b, measure continuity to the rj45 plug, and maybe even to the dmx out plug on your fixture. I'd also double check the ground and the address/settings of the fixture. Lastly you can only try a terminating resistor and make sure that the cable isn't too long. Oh, you did doublecheck your pin assignments, right, both in the dmx bus declaration and the uart?

But debugging this setup probably goes far beyond what's useful in this Github issue - so maybe drop by on the esphome discord, and we can discuss more in depth with more code and more pictures.

johanneswilkens, 2021-12-27:
This is perfect! It's exactly what I need, one thing I noticed, when I used a 6 channel led par (R, G, B, UV, Dimmer, Strobe) all 6 channels need to be defined in the dmx component otherwise the fixture wont work. Other than that less than 5 minutes to set up with ESP32 and MAX3485, would be nice to be able to use it on UART1 of ES8266 as well since only TX is required.
Now it's time for light components that support dimmer, strobe effect, rgba/rgb-uv/rgbaw/rgbaw-uv (or any combo of the previous channels to use any professional light fixture) or maybe even color wheel and xy position (for moving lights like moving heads and scanners)

andyboeh, 2022-01-10:
@johanneswilkens I think that a port to the ESP8266 is doable, but I suspect that the frame rate will be limited.
Next week, I should receive a new DMX dimmer and an RS485 adapter board, I can try to come up with some code.

jakosek, 2022-01-10:
@andyboeh try some low level code from my first attempt on 8266:
`void DMX512::sendBreak() {
   SET_PERI_REG_MASK(UART_CONF0(SEROUT_UART), UART_TXD_BRK);
   delayMicroseconds(DMX_BREAK);
   CLEAR_PERI_REG_MASK(UART_CONF0(SEROUT_UART), UART_TXD_BRK);
   delayMicroseconds(DMX_MAB);
}`

andyboeh, 2022-01-11:
Thanks, I'm going to do that!

johanneswilkens, 2022-01-11:
Would be cool, if needed I can test it with a string of multiple rgb fixtures to see how it handles more fixtures/channels.

andyboeh, 2022-01-11:
I had a bit of time today and added completely untested support for the ESP8266 in the esp8266_support branch. Your yaml needs to look like this for testing it:
```
external_components:
  - source: github://andyboeh/esphome-dmx512@esp8266_support
```

It should work with the hardware UARTs 0 and 1.

nagyrobi, 2022-01-13:
@andyboeh can you please suggest a yaml config to set up the ADJ VBAR PAK as an RGB light? It also has some effects, can those be added too?

andyboeh, 2022-01-13:
I'm sorry, I can't help with specific devices that I don't know/have. You have to look at the manual of the device, figure out which channels correspond to which parameter and configure it accordingly.

johanneswilkens, 2022-01-13:
@nagyrobi this is for a home brand par from thomann, i think this might help you.

```
external_components:
  - source: github://andyboeh/esphome-dmx512

uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: 5
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  enable_pin: 33
  tx_pin: 5
  uart_num: 1

output:
  - platform: dmx512
    channel: 1
    universe: dmx
    id: dmx_1
  - platform: dmx512
    channel: 2
    universe: dmx
    id: dmx_2
  - platform: dmx512
    channel: 3
    universe: dmx
    id: dmx_3
  - platform: dmx512
    channel: 4
    universe: dmx
    id: dmx_4
  - platform: dmx512
    channel: 5
    universe: dmx
    id: dmx_5
  - platform: dmx512
    channel: 6
    universe: dmx
    id: dmx_6

light:
  - platform: rgb
    name: "SePar RGB"
    red: dmx_1
    green: dmx_2
    blue: dmx_3
    id: RGB
    
  - platform: monochromatic
    name: "SePar UV"
    id: UV
    output: dmx_4

  - platform: monochromatic
    name: "SePar Dimmer"
    id: dimmer
    #internal: true
    output: dmx_5

  - platform: monochromatic
    name: "SePar Strobe"
    output: dmx_6
```

andyboeh, 2022-01-14:
> Can you point me in the right direction please? In your [repo](https://github.com/andyboeh/esphome-dmx512/tree/esp8266_support) I only see config example for the main component, but no example how to configure a dimmer relying on it.

The complete example is in the file `example_dmx.yaml`, not in the README: https://github.com/andyboeh/esphome-dmx512/blob/esp8266_support/example_dmx.yaml

nagyrobi, 2022-01-14:
Ah ok, didn't notice that, thank you.

nagyrobi, 2022-01-16:
> I'm leaving the RE and RO floating.

I connect both RE and RO to +Vcc and it works.

[image]

```
MAX485-M RE & DE -> ESP32 +5V
MAX485-M VCC     -> ESP32 +5V
MAX485-M GND     -> ESP32 GND
MAX485-M DI      -> ESP32 GPIO5
MAX485-M A       -> XLR 3
MAX485-M B       -> XLR 2
MAX485-M GND     -> XLR 1
```

nagyrobi, 2022-01-16:
> only problem is that the dmx address is hard coded, so maybe its possible to change that too

Seems like channel calculation based on address is not supported but can be worked around with [manual calculation](https://github.com/andyboeh/esphome-dmx512/issues/3#issuecomment-1013850969). You jut have to keep in mind when you set up your light fixtures that the next free address number on the bus can be the address of the previous light + the last channel number of that light.

So for example the [ADJ VBar Pak](https://d295jznhem2tn9.cloudfront.net/ItemRelatedFiles/8659/vbar_pak.pdf) can be used in various modes, if you set them up to 8-channel mode (to access all the effects) you can set the address of the first one 1, the next one to 9, the third one to 17, fourth one to 25, using the buttons on each fixture.

Then in the config you'd have channels for the first one from 1 to 8, the second one from 9 to 16, third one from 17 to 24, forth one from 25 to 32.

andyboeh, 2022-01-16:
According to @nagyrobi my DMX component also works on the ESP8266. I pushed support for it to to the main branch and I'd be happy to hear your feedback!

nagyrobi, 2022-01-16:
[image]

nagyrobi, 2022-01-17:
[image]
[image]
[image]
[image]

andyboeh, 2022-01-18:
@nagyrobi fyi: The 120 Ohm resistor is already on the board (R7), at least according to the schematic (https://protosupplies.com/product/max485-ttl-to-rs-485-interface-module/)

johanneswilkens, 2022-01-18:
Maybe also interesting to know for some people, there is also a 3v3 version of the max485 called max3485, so that one would be better for esp, also if you want to use the esp as slave for example then over time you won’t blow out the esp pins with 5V.

andyboeh, 2022-01-18:
Yes, very true! I use the SP3485 which is the equivalent of Sipex.

nagyrobi, 2022-01-18:
I use the MAX485-M at 3.3V only and it works (on another modbus device too).

nagyrobi, 2022-01-18:
> The 120 Ohm resistor is already on the board (R7)

Hm indeed. My previous experience a few weeks ago showed though that MODBUS communication CRC errors happen if I don't add the 120 Ohm resistor myself in parrallel with the connector.

andyboeh, 2022-01-18:
> I use the MAX485-M at 3.3V only and it works (on another modbus device too).

It's specified at 5V, so there is no guarantee that it works on 3.3V. Use the 3485 to be on the safe side.

> Hm indeed. My previous experience a few weeks ago showed though that MODBUS communication CRC errors happen if I don't add the 120 Ohm resistor myself in parrallel with the connector.

Yeah, the exact value depends on your cables and your environment.

nagyrobi, 2022-01-18:
@andyboeh this works so well.
It's evrn possible to combine the various lights into partitions as `single_light_id`s and use them in sync, even with an 8266.

It's worth making this component as part of ESPHone @jesserockz

jesserockz, 2022-01-19:
@nagyrobi if you have it running you should PR that component into ESPHome. I have no objections (unless the code is not up to scratch)

andyboeh, 2022-01-21:
@nagyrobi @jesserockz I consider my DMX implementation a hack. The user has to configure the TX GPIO a second time and needs to know which UART number is in use. To implement that properly, changes to the Arduino libraries are necessary (see #1494).

nagyrobi, 2022-01-21:
It's a nice workaround and does nothing illegal IMHO.

andyboeh, 2022-02-25:
@jesserockz I am considering creating a PR myself. Do you think it's worth creating one even though the UART number and TX pin have to be configured twice?

nagyrobi, 2022-02-25:
What picture?
I took all the photos myself...

nagyrobi, 2022-02-25:
I'll make one

nagyrobi, 2022-02-25:
Here you go:
[image]

Ajwardxxx, 2022-12-05:
Hi guys I have an entire house with lighting based on DMX512, and I know my way around Arduino, C++  and Python. I'm also a programmer of many years experience. I'm fairly new to Home Assistant (just moved over from LinuxMCE), but I'm keen to help get this Esphome solution working, if I can. It seems to have stalled, because @andyboeh doesn't want to support what he considers a hack - fair enough. What can I do?

nagyrobi, 2022-12-05:
**jesserockz**, as one of the main devs from ESPHome is way too overwhelmed. The proper way to get stuff to his table is to prepare a normal PR (eventually marked as draft) + docs (there are many of us who can help with that). Developing related questions can be asked on Discord where several dev contributors can offer support. When done with all of this, PR can be marked for review and Jesse will put it on his todo list. Which, often is about several months long, but it's worth the patience.

jazzmonger, 2023-02-04:
any progress on this project?? I have a fairly straightforward DMX setup (lights, lasers & dimmers)  and want to go wireless.  this seems like it would fit the bill perfectly.

[9 more replies]