esphome/feature-requests #320: Python 3 Migration Plan

OttoWinter, 2019-07-10:
Platformio core v4 has been released: https://docs.platformio.org/en/latest/history.html

As has been said before, I'd like to move to python 3 asap - dealing with unicode in python 2 is just fundamentally broken and python 2 really should not be used anymore.

So the plan is this:

- ESPHome v1.14:

 - Require platformio>=4.0.0
 - Make sure ESPHome works properly with py3 (it should already, but some more testing)
 - Still allow py2.7 and use python2 for all guides/docker image
 - Tell users to try using esphome with python 3 in changelog

Between 1.14 and 1.15:

 - Wait for user feedback. Does anything break under python 3?

- ESPHome v1.15:

 - Require python>=3.5
 - Update docker/hass.io images for python3
 - Potentially: publish dummy python2 esphome package that tells users that they should migrate to python3 (without this, any time `pip2 install -U` is run, pip will correctly not install the python3 version; but then the user never knows about py3 migration and misses all updates)

Potential future work/discussions:

 - With py3, we have async/await coroutine syntax. Maybe we should change esphome's `yield` coroutine syntax to that (depends on how easy that is)

 - Maybe we should go straight to py3.6 - HA will soon(or already does, idk) requires 3.6: https://github.com/home-assistant/architecture/issues/167

  - Anything we need from py3.6? I only see f-strings, which are probably not important enough to require 3.6 over 3.5

brandond, 2019-07-10:
3.5 is already scheduled to go EOL mid-2020 and 3.6 is pretty easy to get a hold of, so if you're going to require a newer version you might as well go for that and save having to deprecate 3.5 in another year.

OttoWinter, 2019-07-10:
> 3.6 is pretty easy to get a hold of

As has been shown in the HA discussions, it's not so easy - for debian systems you need to use the very latest buster dist release, and many people just don't have that release yet. I'd like to make the py3 migration as simple as possible, if py3.6 would be more difficult for users that would count for me as a big minus.

Also, while 3.5 will go EOL soon I don't see a big reason we need any of the 3.5 support releases - it's not like we're handling anything really security-relevant in the python codebase which would need a fast patch update. If we ourselves don't need any 3.6 features, the deprecation of py3.5 should be handled by the distribution, not ourself.

MatthewFlamm, 2019-07-12:
I think that this is one of the situations where the vast majority of people that have no problem upgrading are silent.  I installed py3.7 on raspbian stretch with absolutely no problem. Given that, I assume, a large percentage of ESPHome users use HA, all of those users will be upgrading python in one month unless they stop upgrading. I agree however that any hurdle to use can be too much.

minsuke, 2019-08-14:
You great job, I always pulling for you thx Otto

closed 2019-12-07 by OttoWinter