esphome/feature-requests #325: weekly scheduler for climate

blademckain, 2019-07-15:
**Describe the problem you have/What new integration you would like**
My need is to create a chronothermostat with esp32, a temperature sensor and a thermostatic valve.
I would need a weekly scheduler to be able to set the weekly thermostat temperature.
Similar to this: https://www.smart-thermostat.eu/

**Please describe your use case for this integration and alternatives you've tried:**
I would like the esp32 device to be independent from the rest of the system but at the same time to be configurable from homeassistant(mqtt).
For example, you could put an esp32 in each room and from Home Assistant change the weekly temperature of all esp32. In this way if homeassistant node go down, other esp32 can continue to function autonomously.
the main problem is how to manage the weekly schedule.
A solution could be to use the time component (on_time). But I don't know how to modify it from the outside (homeassistant (mqtt or api))
It would be nice to have a dedicated component, configurable in this way:

` schedule:
      id: my_schedule                                                            
      - weekdays: "1,2"
      -  start: "07:30"
      -  end: "23:00"
         action:
               controller: my_bang_bang
               default_target_temperature_low: 20 °C
               default_target_temperature_high: 22 °C
`
the component should be able to be modified via api or mqtt

glmnet, 2019-07-15:
This is one of these issues where what is requested is out of scope of ESPHome mission.
Have you considering using Home Assistant as scheduler (or another controller)
IMHO that should be a better approach for you.  

Home Assistant can go down, but that is ok, it should not be down long enough so a schedule is missed, e.g. what temperature to use during this whole week.

blademckain, 2019-07-16:
Certainly the main controller would be homeassistant, but in some cases it is essential that the derived devices remain functional even when the central node fails.
And this is one of the features of esphome https://esphome.io/guides/automations.html?highlight=network#do-automations-work-without-a-network-connection

The on_time functionality already exists, it would be enough to make it possible to modify it through api / mqtt

OttoWinter, 2019-07-21:
I will close this because it's already possible with `on_time` - it's correct the values for on_time can't be changed over automations, but the proposed scheduler also can't do that and adding such a feature would not be simple -> won't happen.

I'm not really against the idea of a generic scheduler, but I'd like to wait for the [HA scheduler component](https://github.com/home-assistant/architecture/issues/148) first and see what can be learnt from their experience.

closed 2019-07-21 by OttoWinter