esphome/feature-requests #331: Ability to specify OTA IP on command-line

Joshfindit, 2019-07-17:
**Describe the problem you have/What new integration you would like**
To be able to program an ESP OTA from the command-line without changing the config

Like: `esphome --use_address 192.168.0.2 lamp1.yaml run`

**Please describe your use case for this integration and alternatives you've tried:**
Used mainly for "duplicating" configs
I have used `use_address` as per #100, and it works, but then it's: duplicate file, modify name, modify wifi config, flash, modify wifi config back where it could be just duplicate, modify name, flash.

**Additional context**
Related to https://github.com/esphome/feature-requests/issues/100 but not exactly the same

glmnet, 2019-07-17:
You can do that the syntax is
`esphome file.yaml run --upload-port ip-address`

closed 2019-07-17 by glmnet

Joshfindit, 2019-07-17:
Awesome, thank you.
Apologies for opening the request: I see it now where it was in the documentation