esphome/feature-requests #332: Upload firmware bin through CLI

angustrau, 2019-07-19:
**Describe the problem you have/What new integration you would like**

An additional flag to `esphome upload` that allows specifying a different bin file to upload.

**Please describe your use case for this integration and alternatives you've tried:**

Will allow uploading firmware other than esphome leveraging the OTA mechanism. A similar feature exists in the web server component, but some devices may not have that enabled.

**Additional context**

From what I can tell, it would just need [an extra parser argument](https://github.com/esphome/esphome/blob/7210ad7ed9b1a00c69a5e19231e10923df508418/esphome/__main__.py#L434) and a change to the [argument passed to espota2](https://github.com/esphome/esphome/blob/7210ad7ed9b1a00c69a5e19231e10923df508418/esphome/__main__.py#L191)

Anonym-tsk, 2019-07-23:
Just use [espota.py](https://github.com/esp8266/Arduino/blob/master/tools/espota.py)

mnoorenberghe, 2020-04-05:
I was going to ask if a PR would be accepted for this exact thing. In my case I made the .bin with esphome on other network but need to flash it at a remote location. I had to write a two line python script as a workaround:

```python
from esphome import espota2
espota2.run_ota("my_device.local", 8266, "my_ota_password", "firmware.bin")
```

> Just use [espota.py](https://github.com/esp8266/Arduino/blob/master/tools/espota.py)

Is there any downside to this? Why doesn't esphome use that code if it's equivalent?