esphome/feature-requests #333: Climate add FAN and other modes

glmnet, 2019-07-19:
**Describe the problem you have/What new integration you would like**

As HA 0.96 introduced the Climate component refactor, may be it is time to add a few more features to the ESPHome Climate integration.
Here are the [Climate docs](https://developers.home-assistant.io/docs/en/entity_climate.html)
I'd like to see implemented:
STATE_DRY, STATE_FAN_ONLY
SUPPORT_FAN_MODE, SUPPORT_SWING_MODE

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

These all can be coded to IR Remotes

It seems the Climate component needs a few bug fixes yet, 0.96 is not working with ESPHome, I did not fill a bug as there were many Climate related bugs filled already.

MiguelAngelLV, 2019-07-22:
The HA 0.96 compatibility works fine with 0.96.3

The DRY and FAN_ONLY states and SWING suppport I need it too.

glmnet, 2019-09-23:
So, I started working on this, don't expect something quick as this requires me to learn more Python, Home Assistant, ESPHome API based on Google Protocol Buffers, mqtt, etc. It should not be too hard though.

closed 2019-11-16 by OttoWinter