esphome/feature-requests #337: Support for multiple i2c busses

tdamsma, 2019-07-21:
**Describe the problem you have/What new integration you would like**
I would like to connect multiple sensors with a fixed I2C address

**Please describe your use case for this integration and alternatives you've tried:**
I have multiple bme280 that I want to connect to the same esp32. As there are plenty of pins capable of i2c I managed to do this with micropython by just setting up multiple i2c busses, no problem. With esphome there is no support for that

**Additional context**
Found a similar (misfiled) feature request here: https://github.com/esphome/issues/issues/473

OttoWinter, 2019-07-21:
Already possible:

```yaml
i2c:
- sda: GPIO0
  scl: GPIO2
  id: i2c_bus1
- sda: GPIO3
  scl: GPIO4
  id: i2c_bus2

sensor:
- platform: bme280
  i2c_id: i2c_bus1
  # ...
```

closed 2019-07-21 by OttoWinter

tdamsma, 2019-07-21:
@OttoWinter, great! thanks. I did try to google and read docs, but didn't find this mentioned anywhere.