esphome/feature-requests #339: Dynamically adjust deep sleep component's configuration

jeff-h, 2019-07-22:
**Describe the problem you have/What new integration you would like**

It would be useful to be able to dynamically adjust the deep sleep component's configuration. 

**Please describe your use case for this integration and alternatives you've tried:**

For example, when solar / battery powered, it would be useful to have it sleep a lot during the evening, and sleep less during the day.

**Additional context**

https://github.com/esphome/issues/issues/528

OttoWinter, 2019-10-13:
Yes that's already possible through the use of the `deep_sleep.enter` action.

Set up the `deep_sleep` component with a really long run_duration (30days for example) so that won't trigger the deep sleep.

Then, you can trigger the deep_sleep manually in any action with a call to `deep_sleep.enter`.

stephenmhall, 2019-10-20:
I think I am missing something here, would calling deep_sleep.enter just start the 30 day timer? My use for an adjustable deep_sleep timer is for a light sensor that would take readings every 9 minutes through the day but once it hit 0.0 it would switch to 60 minutes to conserve the battery through the night. When I wrote my own code I got the sunset sunrise times from my server and used that to determine the length of deep sleep.

OttoWinter, 2019-10-20:
@stephenmhall 

> This action makes the given deep sleep component enter deep sleep **immediately**.

stephenmhall, 2019-10-20:
Aye, I understand that, but would it not go into the 30 day deep sleep that was initially set, pretty much rendering it dead? The enter and prevent actions are fine to keep the device awake to allow OTA programming or as in the solar case to keep it on permanently until the batteries started going down, but you would then have to deep_sleep.enter into your standard deep sleep cycle, not 30 days but the 9 minutes or whatever is best. What myself and jeff-h were after was a deep-sleep duration dependent on some variable like r-jordan did in issue #528 by calling setup, which obviously is not a good idea. Also thanks for the super fast reply :)

OttoWinter, 2019-10-20:
Ah, now i understand. Yes that is indeed an issue

ignisf, 2019-10-29:
Hello, @OttoWinter, I might try my hand at a PR for this. Off the top of your head, what would be a good direction/approach for implementing this? Maybe make run and sleep durations templatable?

OttoWinter, 2019-10-30:
I would recommend doing it this way:

1. Add a new option `sleep_duration` to the `deep_sleep.enter` action - any value here will override the `deep_sleep` component's configuration when called at runtime.

2. Make `sleep_duration` and `run_duration` optional fields in the main config. If they're not present, they have default values of infinite (or whatever the max value of the datatype is).

jappish84, 2020-01-09:
+1 for this feature. Would be very useful to be able to set sleep_duration dynamically, say if you want the device to wake up at 22.32pm, you wake up once, check time, set new sleep_duration and wake up only then, instead of a static time interval

ntompson, 2020-04-05:
+1. I have two things in mind for this:

- Increase sleep duration during the evening as already discussed
- Increase sleep duration during the day if battery voltage is unusually low

The second point is important to me at the moment - I've got a solar / battery powered ESP32 working well in most conditions, but I'm finding that if I get a few days of rain, there isn't enough solar to keep the battery up and the battery drains to the point of no return. I could just design a larger solar / battery, but some simple smarts could preserve the battery on the odd occasion the weather is unusually poor.

TheHomieFox, 2020-04-30:
+1, this would be a great feature to have.

crzykidd, 2020-05-20:
+1 and/or read deep sleep time from MQTT and use that for next deep sleep.

pat1300, 2020-05-22:
+1 : very interested

Bierchermuesli, 2020-05-22:
+1
Workaround: I use a simple NodeRed Flow which puts the esp to sleep for a hour (or every time it wakes up and conditions aren't met) - depending on (Solar Power, Weather at Night, etc...)

```
deep_sleep:
    id: do_sleep
    sleep_duration: 1h
    run_duration: 30days
switch:
  - platform: template
    name: "Send to sleep"
    turn_on_action:
      - deep_sleep.enter: do_sleep
```
and NR:
```
[{"id":"bcfa473d.fe5f28","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"222743b2.91aafc","type":"trigger-state","z":"bcfa473d.fe5f28","name":"on Wakeup","server":"8d33c7a.8b9b338","exposeToHomeAssistant":false,"haConfig":[{"property":"name","value":""},{"property":"icon","value":""}],"entityid":"binary_sensor.rabattli_esp1_status","entityidfiltertype":"exact","debugenabled":false,"constraints":[{"id":"qd5z7911g9","targetType":"this_entity","targetValue":"","propertyType":"current_state","propertyValue":"new_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"on"},{"id":"jpln3587fh8","targetType":"this_entity","targetValue":"","propertyType":"previous_state","propertyValue":"old_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"off"}],"constraintsmustmatch":"all","outputs":2,"customoutputs":[],"outputinitially":false,"state_type":"str","x":890,"y":340,"wires":[["9cd60f75.21fde"],[]]},{"id":"ea5c1dcb.4ea8d","type":"api-call-service","z":"bcfa473d.fe5f28","name":"send to deepsleep","server":"8d33c7a.8b9b338","version":1,"debugenabled":false,"service_domain":"switch","service":"turn_on","entityId":"switch.deepsleep","data":"","dataType":"json","mergecontext":"","output_location":"","output_location_type":"none","mustacheAltTags":false,"x":1570,"y":380,"wires":[["49e0a370.3a36dc"]]},{"id":"9cd60f75.21fde","type":"time-range-switch","z":"bcfa473d.fe5f28","name":"","lat":"47.0031","lon":"7.2736","startTime":"sunset","endTime":"sunrise","startOffset":"-1","endOffset":"+1","x":1180,"y":340,"wires":[["da1e902f.6743b"],[]]},{"id":"da1e902f.6743b","type":"delay","z":"bcfa473d.fe5f28","name":"","pauseType":"delay","timeout":"20","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":1380,"y":320,"wires":[["ea5c1dcb.4ea8d"]]},{"id":"49e0a370.3a36dc","type":"api-call-service","z":"bcfa473d.fe5f28","name":"Log","server":"8d33c7a.8b9b338","version":1,"debugenabled":false,"service_domain":"logbook","service":"log","entityId":"binary_sensor.rabattli_esp1_status","data":"{\"name\":\"Rabattli1\",\"message\":\"Go To sleep via NR by {{topic}}\"}","dataType":"json","mergecontext":"","output_location":"payload","output_location_type":"msg","mustacheAltTags":false,"x":1810,"y":380,"wires":[["d8bc6e7e.be7e9"]],"icon":"font-awesome/fa-list"},{"id":"d74bb1af.c8356","type":"time-range-switch","z":"bcfa473d.fe5f28","name":"","lat":"47.0031","lon":"7.2736","startTime":"sunrise","endTime":"sunset","startOffset":"+1","endOffset":"-1","x":1180,"y":440,"wires":[["ea5c1dcb.4ea8d"],[]]},{"id":"a85e259a.820258","type":"trigger-state","z":"bcfa473d.fe5f28","name":"OnDischarge","server":"8d33c7a.8b9b338","exposeToHomeAssistant":false,"haConfig":[{"property":"name","value":""},{"property":"icon","value":""}],"entityid":"sensor.battery_current","entityidfiltertype":"exact","debugenabled":false,"constraints":[{"id":"z0luyh0o19k","targetType":"this_entity","targetValue":"","propertyType":"current_state","propertyValue":"new_state.state","comparatorType":">","comparatorValueDatatype":"num","comparatorValue":"0.004"}],"constraintsmustmatch":"all","outputs":2,"customoutputs":[],"outputinitially":false,"state_type":"num","x":910,"y":440,"wires":[["d74bb1af.c8356"],[]]},{"id":"d8bc6e7e.be7e9","type":"debug","z":"bcfa473d.fe5f28","name":"","active":true,"tosidebar":true,"console":false,"tostatus":true,"complete":"topic","targetType":"msg","x":2040,"y":380,"wires":[]},{"id":"a67012fe.09124","type":"inject","z":"bcfa473d.fe5f28","name":"at 23:00","topic":"SleepAtNight","payload":"","payloadType":"date","repeat":"","crontab":"00 23 * * *","once":false,"onceDelay":0.1,"x":900,"y":240,"wires":[["ea5c1dcb.4ea8d"]]},{"id":"d99c907c.eed9b","type":"comment","z":"bcfa473d.fe5f28","name":"Info","info":"Puts the ESP into deeps sleep for one hour\n\nTriggered by\n1. Low Solar Energy durin day\n2. latest at 23:00 Clock\n3. after every wakup during night","x":870,"y":200,"wires":[]},{"id":"8d33c7a.8b9b338","type":"server","z":"","name":"your-server","legacy":false,"addon":false,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true,"cacheJson":true}]
```

Of course it would be nice to have a `deep_sleep.enter` function with time-duration as argument

dircr, 2020-05-27:
+1 for this

I have an oil-level sensor that I would like to report twice a day if I have >300L in the tank and perhaps every 30 minutes if it's less than  100L.

ofalvai, 2020-06-07:
I ended up writing a custom ESPHome component that reads sleep duration from an MQTT topic and goes to sleep when another MQTT topic receives a message: https://gist.github.com/ofalvai/4d5b6fd4a7fd31f2495f2846464d4500

rt87, 2020-06-12:
+1
My usecase: LED nightlights are only relevant during the night. I could of course use ambient light sensors, but I would prefer to just go by the official dusk/dawn times (+/- an offset, if necessary) for all my ESPs. So once a day these times should be communicated and then the ESPs would go to sleep at a given time and for a given duration, which of course both vary depending on the exact date/season.

quenbert, 2020-08-06:
+1 too. About the implementation, isn't possible to make the run_duration & sleep_duration parameters templatable?
My use case: only sleep at night, between sunset and sunrise.

quenbert, 2020-08-30:
Hi all interested. In fact, after a bit of digging, it's already possible to update the sleep-duration through a lambda call:
`
deep_sleep:
  id: your_deep_sleep
`
then later in any relevant trigger (I used a time/on_time trigger):
`
- lambda: |-
                    id(your_deep_sleep).set_sleep_duration(YOUR_DURATION_IN_MS);
`

rt87, 2020-09-24:
> Hi all interested. In fact, after a bit of digging, it's already possible to update the sleep-duration through a lambda call:
>
> ```
> deep_sleep:
>   id: your_deep_sleep
> ```
>
> then later in any relevant [...]

Thx! So what would be the best way to make a system such as HomeAssistant tell the ESPHome every day its new sleep times?

Sry, I'm not that experienced with these triggers and such.

quenbert, 2020-09-24:
> Thx! So what would be the best way to make a system such as HomeAssistant tell the ESPHome every day its new sleep times?

The way I did it: in HomeAssistant (not in EspHome config file), define 2 time helpers (Configuration->Helpers), for example:
- chicken_param_hibernate_time
- chicken_param_wake_up_time

These helpers will allow you to setup the times easily in HA lovelace interface. Then in your HA configuration.yaml, define 3 template sensors:

    sensor:
      - platform: template
        sensors:
          chicken_param_hibernate_hour:
            value_template: '{{ (states.input_datetime.chicken_param_hibernate_time.attributes.timestamp | int) | timestamp_custom("%H", false)}}'
            unit_of_measurement: 'h'
          chicken_param_hibernate_minute:
            value_template: '{{ (states.input_datetime.chicken_param_hibernate_time.attributes.timestamp | int) | timestamp_custom("%M", false)}}'
            unit_of_measurement: 'min'
          chicken_param_hibernate_duration_sec:
            # Note: the 24*60*60 is to add 24h, as my use case was to make the esphome device sleep over night, and wake up in the next morning
            value_template: '{{ (states.input_datetime.chicken_param_wake_up_time.attributes.timestamp | int) + (24*60*60) - (states.input_datetime.chicken_param_hibernate_time.attributes.timestamp | int) }}'
            unit_of_measurement: 's'

Now, in your ESPHome configuration, use the sensors exported from HomeAssistant

    sensor:
      # External config from home assistant
      - platform: homeassistant
        name: "Chickenbox Hibernation hour from HA"
        id: chickenbox_hibernation_hour_from_ha
        entity_id: sensor.chicken_param_hibernate_hour
      - platform: homeassistant
        name: "Chickenbox Hibernation minute from HA"
        id: chickenbox_hibernation_minute_from_ha
        entity_id: sensor.chicken_param_hibernate_minute
      - platform: homeassistant
        name: "Chickenbox Hibernation duration from HA"
        id: chickenbox_hibernation_duration_from_ha
        entity_id: sensor.chicken_param_hibernate_duration_sec
            
Define your sleep:

    # Deep sleep aka hibernation
    deep_sleep:
      id: chickenbox_deep_sleep

And define an on_time trigger to enter deep sleep a.k.a hibernation:

    time:
      - platform: homeassistant
        id: ha_time
        on_time:
          - seconds: 0
            then:
            # Trigger the hibernation
            - if:
                condition:
                  - lambda: |-
                      return (id(ha_time).now().is_valid() && id(ha_time).now().hour == id(chickenbox_hibernation_hour_from_ha).state && id(ha_time).now().minute == id(chickenbox_hibernation_minute_from_ha).state);
                then:
                  - logger.log: "Time to sleep !!! Hibernating"
                  - lambda: |-
                      id(chickenbox_deep_sleep).set_sleep_duration(id(chickenbox_hibernation_duration_from_ha).state * 1000);
                  - delay: 1s
                  - deep_sleep.enter: chickenbox_deep_sleep
                  
Note that this method will trigger ESPHome every minute to check if it matches the triggering conditions, but in practice, it works...

Hope I've not messed the code in my different copy/paste. And for info, the 'chicken' prefix is because of the application: to pilot the door of my chicken coop.

fabiosoft, 2021-01-29:
Hi, i am trying to stop a binary sesor to change state for 1h after i click a button on GPIO
so i tought about put the device to deep sleep and it would be nice to have a `deep_sleep.enter` function with time-duration as argument.

Any updates on this?

closed 2021-02-15 by glmnet