esphome/feature-requests #341: TTGO T-Camera ESP32 Face recognition

thundergreen, 2019-07-22:
**Describe the problem you have/What new integration you would like**

I'd like to know if it's possible to add face recognition to this cam
**Please describe your use case for this integration and alternatives you've tried:**

There's an official Arduino Sketch which supports face recognition and I'd like to see this in esphome. Is it possible to add this to have some automation running this?

tpbosolo, 2019-10-06:
Indeed, face recognition would be very useful within esphome. Any updates?

OttoWinter, 2019-10-20:
Nope, this is unfortunately not possible - ESPHome puts the camera in a mode where it only receives data in JPEG form and streams it directly to HA.
There's not enough resources to then also decode JPEG and store it there for processing (without putting the camera in a different mode).

Camera processing is not really a direction ESPHome should go in, it gets way too hairy (someone suggested barcode reading, QRCode reading, now face recognition). All those things should be done in HA on a machine with way more resources.

closed 2019-10-20 by OttoWinter