esphome/feature-requests #343: Use of light.dim_relative expressed in lambdas

Adorem, 2019-07-23:
Hi,

**Describe the problem you have/What new integration you would like**

According to these two issues you recommend to use the new `light.dim_relative`. Then I would like to know if it’s possible to use it expressed in lambdas?

https://github.com/esphome/feature-requests/issues/185#issuecomment-486768256
https://github.com/esphome/issues/issues/347

**Please describe your use case for this integration and alternatives you've tried:**

On the same device I have LED strip and IR Receiver.
I would like not to expose all the button of my remote to HA and want to directly interact with the light. For exemple the following works fine for the on/off buttons. But now I would like something for the + and - buttons.

```YAML
light:
  - platform: rgb
    name: "LED"
    red: output_component1
    green: output_component2
    blue: output_component3
    id: led_strip

output:
  - platform: esp8266_pwm
    id: output_component1
    pin: GPIO5
  - platform: esp8266_pwm
    id: output_component2
    pin: GPIO13
  - platform: esp8266_pwm
    id: output_component3
    pin: GPIO12

remote_receiver:
  pin:
    number: GPIO04
    inverted: True
  dump: all
  on_nec:
    then:
      - lambda: |-
          if( 0xFF == x.address  )
          {
            switch( x.command )
            {
              case 0x906F: // dimmer +
                {
                  float brightness;
                  auto call = id(led_strip).turn_on();
                  id(led_strip).current_values_as_brightness( &brightness );
                  brightness += 0.1f;
                  if( brightness > 1.0f )
                  {
                    brightness = 1.0f;
                  }
                  call.set_brightness( brightness );
                  call.perform();
                }
                break;
              case 0xB847: // dimmer -
                {
                  float brightness;
                  auto call = id(led_strip).turn_on();
                  id(led_strip).current_values_as_brightness( &brightness );
                  brightness -= 0.1f;
                  if( brightness < 0.0f )
                  {
                    brightness = 0.0f;
                  }
                  call.set_brightness( brightness );
                  call.perform();
                }
                break;
              case 0xF807: // power off
                {
                  auto call = id(led_strip).turn_off();
                  call.perform();
                }
                break;
              case 0xB04F: // power on
                {
                  auto call = id(led_strip).turn_on();
                  call.set_brightness(1.0);
                  call.set_rgb(1.0, 1.0, 1.0);
                  call.perform();
                }
                break;
              default:
                break;
            }
          }

```

Even for this code, by using the current_values_as_brightness(), I probably made some errors because it's also not working.

**Additional context**

Thanks,
Christophe

OttoWinter, 2019-10-13:
No, there's not a direct `dim_relative` call, but with the existing `call.set_brightness` you already have everything you need.

For reference, here's the implementation for `dim_relative` https://github.com/esphome/esphome/blob/5c9755ecc161e43ce0550148b2897d9ae565d8ba/esphome/components/light/automation.h#L68-L77 (basically just adds a value to the current brightness and commits that through set_brightness)

closed 2019-10-13 by OttoWinter

Adorem, 2019-10-14:
Thanks, in my previous code I was doing this:
`id(led_strip).current_values_as_brightness( &brightness );`
But according to the code you show me I shall do this:
`id(led_strip).remote_values.as_brightness( &brightness );`

And now with this change it works has expected. Thank you again.