esphome/feature-requests #345: Add climate remote component for Fujitsu AC unit

r-jordan, 2019-07-24:
**Describe the problem you have/What new integration you would like**

Add new climate remote component to control a Fujitsu AC unit by sending IR signals.

**Please describe your use case for this integration and alternatives you've tried:**

There are other similar components created right now and new ones are being developed (Coolix, Tcl112, Yashima). This component is just another one with the same functionality.

**Additional context**

The codes can be obtained from public libraries that control several AC types. For example:

https://github.com/ToniA/arduino-heatpumpir/blob/master/FujitsuHeatpumpIR.cpp

glmnet, 2019-07-29:
I'll try to get this one during this week, are you able to copy some code in the esphome folder?

31337Ghost, 2019-07-29:
Got implemented in PR[#677](https://github.com/esphome/esphome/pull/677) but obviously some general stuff should be refactored.

r-jordan, 2020-01-03:
Hi, I've tested this component in version 1.15.0-dev and it works for me.

Thank you all!

glmnet, 2020-01-03:
Thanks, closing as implemented in https://github.com/esphome/esphome/pull/677

closed 2020-01-03 by glmnet