esphome/feature-requests #346: Add Toshiba AC but possibility of library to cover multiple manufactures

KShips, 2019-07-27:
**Describe the problem you have/What new integration you would like**
Looking for support for Toshiba AC units, have found on Tony https://github.com/ToniA/arduino-heatpumpir/blob/master/ToshibaHeatpumpIR.cpp and if possible have this added.

I have come accross this library that was originally forked from ToniA but for ESP to receive and transmit AC for many manufacturers, looks to be actively developed. Not sure if useful for esphome? https://github.com/crankyoldgit/IRremoteESP8266 

**Please describe your use case for this integration and alternatives you've tried:**
Have tried ToniA version and works well.

**Additional context**

glmnet, 2019-07-28:
Is that one working with your ac?
I can add but I have not the ac so I cannot test

KShips, 2019-07-28:
Thanks, yes this one us currently working with my ac https://github.com/ToniA/arduino-heatpumpir/blob/master/ToshibaHeatpumpIR.cpp

glmnet, 2019-07-28:
Ok. I think I can write an integration based on that code. I’d like @ottowinter to suggest how we can do some kind of beta testing this. 
Are you running esphome as hassio add on or how?

OttoWinter, 2019-07-28:
@glmnet You can share custom components as a folder that the user copies into their `custom_components` folder of their ESPHome install. Similar to HA's custom functionality.

baskraai, 2019-07-28:
I have been working on this all weekend. But I do not get how this works:
```
The example below is an example of a custom climate device - all climate devices must override two methods (Climate ):

traits: This should return a ClimateTraits object representing the capabilities of the climate device.
control: This receives a ClimateCall object that contains the command the user tried to set.
```
If someone can give me an example based on the example of [ESPHome Custom Climate](https://esphome.io/components/climate/custom.html) I can make a multi-unit compatible component. Currently I have a working implementation of the custom light that can turn on and off the AC with the IRremoteESP8266 library.

KShips, 2019-07-28:
@glmnet Yes running on Hassio.

glmnet, 2019-07-28:
At the beginning we were going to have a `model` option on a `ir_climate` platform, e.g.

````yaml
climate:
  ir_climate:
    model: toshiba
````

but then we decided to use a platform per manufacturer approach, that surely makes code more simpler and maintainable and thus we now have `coolix` `tcl112` etc. One of the best features of ESPHome is consistency so I don't think this were going to change on the codebase now.

Also reusing ESPHome integrated IR libraries, ESP8266 and ESP32 are supported, and also receiver can easily be created so you can keep your AC state even when using the original remote transmitter. This is coming here: https://github.com/esphome/esphome/pull/645

Of course you can go on on your own and create a custom component that will re use another library.

If you own an unsupported AC and feel like coding for ESPHome, I suggest you look at ``coolix`` or other of the already implemented equipment and try to get it working, sending IR codes is a lot of fun :)
Create a feature request for the specific model so we don't duplicate effort on the same model.

glmnet, 2019-07-29:
Alright, I've got the toshiba first version. The documentation from coolix applies mostly except you have to put `toshiba` instead of `coolix`

Create a folder named `custom_components` at the same level as your yaml files. Then inside that folder create another folder named `toshiba` then put the files there.

You can grab the files from [here](https://github.com/glmnet/esphome/tree/climate-toshiba/esphome/components/toshiba)

diraimondo, 2019-07-29:
@glmnet: I did this for my yashima AC unit (the related component will appear in the next ESPHome release) but I'm not against the creation on a new generic component that would exploit the (good) work made to develop such generic external libraries to manage different brands AC units. I agree on the necessity to don't break the compatibility of the existent components (one is the mine) but it should not be a big problem.

With such generic component one could catch dozens of brand with one (more involving) effort. 

As side note: my AC unit is not supported by such libraries so I'm a living counter-example on why it is necessary to keep the old components...

KShips, 2019-07-30:
> Alright, I've got the toshiba first version. The documentation from coolix applies mostly except you have to put `toshiba` instead of `coolix`
>
> Create a folder named `custom_components` at the same [...]

Thanks, I've done a quick test and the unit is registering commands. Off command works, still need to check others as I was in a rush, back home end of week so will report back asap. 

Is it sending fan auto and temp either 23 (which look to be hard coded) or what is set in target_temperature in yaml on every command (Auto/Cool/Heat)?
If so, I can add some details to https://github.com/esphome/feature-requests/issues/333 to be able to set to temp for the unit in front end.

baskraai, 2019-07-30:
> Thanks, I've done a quick test and the unit is registering commands. Off command works, still need to check others as I was in a rush, back home end of week so will report back asap.
>
> Is it sending [...]

Nice work, I will add your overrides to my code and check if I can make it multi-platform! So we can have a component that works with multiple AC systems.

glmnet, 2019-07-30:
> Is it sending fan auto and temp either 23 (which look to be hard coded) or what is set in target_temperature in yaml on every command (Auto/Cool/Heat)?

It is sending fan auto as you cannot select fan speed from HA ui yet. Temperature 23 is just an initialization and the value is overridden later, I should remove the 23.

If everything works I'll go ahead and create a PR for this one.

gervaso, 2019-07-31:
Hello, I've added support for the Carrier devices (my Olimpia Splendid uses the same IR codes), adapting the code from TonyA and the work of @glmnet but there are some issues.
It compiles fine using the custom_components, uploads to the esp and registers in HA, I can also see that the 940nm led emits signals (using a smartphone camera) but nothing happens on the climate units. Last year I had a working demo based on TonyA library, the same IR emitter and an arduino, integrated using "mysensors" and I was able to command the conditioners (ON/OFF, HEAT/COOL/AUTO/DRY and the fan speeds).

Maybe I made some mistakes porting the code, could you please check and eventually integrate it?
The reference on TonyA library is CarrierMCAHeatpumpIR in https://github.com/ToniA/arduino-heatpumpir/blob/master/CarrierHeatpumpIR.cpp

My files: https://github.com/gervaso/esphome/tree/dev/esphome/components/carrier

Thank you!

gervaso, 2019-07-31:
> Also reusing ESPHome integrated IR libraries, ESP8266 and ESP32 are supported, and also receiver can easily be created so you can keep your AC state even when using the original remote transmitter. This is coming here: [esphome/esphome#645](https://github.com/esphome/esphome/pull/645)

I'm interested in this because making it "generic" will easily allow us to create systems like the [sensibo](https://sensibo.com) exploiting esphome and the native integration with HA.

glmnet, 2019-07-31:
First of all create a new issue or feature request. Meanwhile...

> My files: https://github.com/gervaso/esphome/tree/dev/esphome/components/carrier

I don't think you can call two times `transmit.perform() `I believe ESPHome caches that data and transmits it when he wants so you might be effectively sending only the later.

Try removing lines from [156](https://github.com/gervaso/esphome/blob/d7548dca254682322ab29b40f5c0f8d9e6cbe87b/esphome/components/carrier/carrier.cpp#L156)-159 and see what happens.

gervaso, 2019-07-31:
> Try removing lines from [156](https://github.com/gervaso/esphome/blob/d7548dca254682322ab29b40f5c0f8d9e6cbe87b/esphome/components/carrier/carrier.cpp#L156)-159 and see what happens.

That made it, thank you! I wasn't aware of the esphome internals and data caching.
I'll cleanup the files and I'll open a PR asap.

KShips, 2019-08-06:
> Alright, I've got the toshiba first version. The documentation from coolix applies mostly except you have to put `toshiba` instead of `coolix`
>
> Create a folder named `custom_components` at the same [...]

Unfortunately I've hit some problems, first one is the Toshiba code doesn't seem to register on the AC or is intermittent. What is confusing is that I accidentally flashed the esp with Coolix and it seems to work fine, maybe they use the same IR signals or I have mixed something up. Doesn't explain why Toshiba code isn't working though.

I will try and put together a IR receiver and compare codes coming from the remote and the esp using Toshiba and Coolix options to see what is going on.
That leads to my other problem, I've just brought another house and will have current house on market asap so I'm not 100% sure if I will get all this sorted. My aim is to at least do some IR code comparisons to see what is going on and will post findings here in case anyone else with a Toshiba picks this up (if easy fix I might get it sorted before I loose access to the old house).

glmnet, 2019-08-06:
Yeah they might be kind of compatible. 
I’m trying to put the codes here. https://github.com/esphome/issues/wiki/AC-Protocols

strumf666, 2019-08-07:
> The reference on TonyA library is CarrierMCAHeatpumpIR in https://github.com/ToniA/arduino-heatpumpir/blob/master/CarrierHeatpumpIR.cpp

This library contains IR codes for many climates including zibro/qlima, which I own; are/will this be supported as well?

glmnet, 2019-08-07:
> This library contains IR codes for many climates including zibro/qlima, which I own; are/will this be supported as well?

The Toshiba seems similar to this https://github.com/ToniA/arduino-heatpumpir/blob/10708306ac715e132283947a96ff927fbc7ee239/CarrierHeatpumpIR.cpp#L129 but I think you just have to try. 
If it works, report back so I can update the compatibility list.
If it doesn't work create an issue or feature request and I can add another model.

strumf666, 2019-08-07:
Ok, thanks. I'll try and report back.

strumf666, 2019-08-10:
Seems I'll need so help to be able to test. I made a folder "custom_components" in esphome folder (where the .yaml files are for each of my esp32 boards) inside I created another folder "toshiba" and I copied the files from [here]([(https://github.com/glmnet/esphome/tree/climate-toshiba/esphome/components/toshiba)]) inside.
I add the following in the yaml of esphome instance:
`climate:
  - platform: toshiba
    name: "Kino_soba klima"
    id: "kino_soba_klima"
    sensor: "kino_temperature"
    visual:
      min_temperature: 17 °C
      max_temperature: 30 °C
      temperature_step: 1 °C`

But I get an error "Platform not found: climate.toshiba"

strumf666, 2020-01-02:
I just tested this with the current version (1.14.3) and it works. I still have the custom components folder as I wrote in previous post, but the config in yaml is this:

`climate:
  - platform: coolix
    name: "Kino_soba klima"
    id: "kino_soba_klima"
    sensor: "kino_temperature"
    visual:
      min_temperature: 17 °C
      max_temperature: 30 °C
      temperature_step: 1 °C`

glmnet, 2020-01-02:
Do you mean, it compiles, or it actually controls your AC? in the later case your AC is using the Coolix protocol not the toshiba 

If you still want to give the toshiba a try check:

1. double check the folder names.
2. how are you getting the files? a common mistake is to right click the file name link and do "Save target as..." this will give you html back, you need to open the file and right click the "Raw" button

strumf666, 2020-02-07:
Sorry for late reply; now it actually controls the AC, first time I tried it didn't do anything, so I am guessing my attempt with custom components did something?

glmnet, 2020-02-07:
Great I’m gonna add Toshiba to dev then

glmnet, 2020-07-25:
fixed in https://github.com/esphome/esphome/pull/1121

closed 2020-07-25 by glmnet