esphome/feature-requests #349: Filter/drop sensor readings outside expected reange

tdamsma, 2019-07-29:
**Describe the problem you have/What new integration you would like**
Sometimes sensors return obviously wrong data. Would be nice to easily filter it out. I have a BME280 that sometimes (once every 100 measurements) returns nonsense (indoor temperature of +160C or -160C). These spikes mess up the graph scaling of the home assistant dashboards, so it would be nice to prevent such spikes from being sent to home assistant in the first place.

**Please describe your use case for this integration and alternatives you've tried:**
I guess I should be able to address this with a lambda statement, but would be nicer to just set a min/max and discard measurements outside of the range

**Additional context**

glmnet, 2019-07-29:
You can also use filters which will look better. 

I wonder if this is a common problem of those sensors or only yours. 

May be a warn could be logged when sensor reports out of sensor range value. What is the range of the sensor?

brandond, 2019-07-29:
yeah I would try a lambda filter: https://esphome.io/components/sensor/index.html#lambda-filter

I believe you can have the lambda `return {}` to drop the value, but I'm not 100% sure on that.

tdamsma, 2019-07-30:
@brandond, thanks! This seems to work (filter out negative values, use -1 as sentinel):

```
    filters:
      - lambda: |-
          if (x >= 0) {
            return x;
          } else {
            return -1;
          }
      - filter_out: -1
```

OttoWinter, 2019-10-13:
Yes, that's also equivalent to:

```yaml
filters:
  - lambda: |-
      if (x >= 0) return x;
      else return {};
```

I think the lambda filter already makes this functionality easy enough, so I will close the feature request - maybe a `range` filter might be useful, but I guess a separate FR would be better specifying exactly how it should work (inclusive ends, syntax etc)

closed 2019-10-13 by OttoWinter