esphome/feature-requests #350: Ability to turn off auto calibration for CO2 sensor MH-Z19B

Spirituss, 2019-07-29:
**Describe the problem you have/What new integration you would like**

At the moment there is no ability to switch off daily auto calibration for the MH-Z19B sensor, while it is switched on by default.
In case it works indoor the measurements it shows can be inadequate due to daily calibration without perfect ventilation. The shift of value can reach 300-400ppm!
The best solution for this case is to switch off auto calibration. "Native" library used in ESPHome as a basis for current sensor support has such an ability with function ```void setAutoCalibration(bool autocalib)```. The manual for library also prescribes:
```If you use this sensor in door, you execute setAutoCalibration(false).```
It is better to give the user ability to choose either he wants to use auto calibration or not.

**Please describe your use case for this integration and alternatives you've tried:**

Sonoff-Tasmota provides support for this sensor with above requested ability:
(https://github.com/arendst/Sonoff-Tasmota/issues/5073)

**Additional context**

Without above described ability the sensor has no real use, since it shows incorrect values.

Animnex, 2019-07-29:
Было бы очень здорово.

to4ko, 2019-07-29:
Please add this function

Iv711, 2019-07-29:
+

Anonym-tsk, 2019-07-31:
Try ESPHome from this branch https://github.com/Anonym-tsk/esphome/tree/mhz19_autocalibration
If it works, I'll make PR.

kooskaspers, 2019-08-04:
Thanks for the efforts @Anonym-tsk, I'll test.
What option needs to be set in the yaml file to disable ABC? Based on your code I'd assume it would be "AUTOMATIC_BASELINE_CALIBRATION: false" right?

Anonym-tsk, 2019-08-04:
Yes, `automatic_baseline_calibration: false`

kooskaspers, 2019-08-04:
And is there any way to force an ABC upon request?

Anonym-tsk, 2019-08-05:
@kooskaspers 
https://github.com/Anonym-tsk/esphome/tree/mhz19_autocalibration
Repo is updated

Actions (call by request):
`mhz19.calibrate_zero`
`mhz19.abc_enable`
`mhz19.abc_disable`

Options (enable or disable ABC on boot):
`automatic_baseline_calibration: true/false`

Example:
```
switch:
  - platform: template
    name: "MH-Z19 ABC"
    optimistic: true
    on_turn_on:
      - mhz19.abc_enable: mhz19sensor
    on_turn_off:
      - mhz19.abc_disable: mhz19sensor

uart:
  rx_pin: 13
  tx_pin: 14
  baud_rate: 9600

sensor:
  - platform: mhz19
    id: mhz19sensor
    co2:
      name: "MH-Z19 CO2 Value"
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 60s
    automatic_baseline_calibration: false
```

I didn't test, because i don't have this sensor %)

kooskaspers, 2019-08-05:
Just cloned your branch:
`git clone --branch mhz19_autocalibration https://github.com/Anonym-tsk/esphome.git`
compiled the binaries:
`sudo ./script/setup`
updated my yaml:

```
esphome:
  name: co2keuken
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "dfadfsafd"
  password: "asfasfdafda"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "dfsaddfdsa"

ota:
  password: "asdfafds"
  
switch:
  - platform: template
    name: "MH-Z19 ABC"
    optimistic: true
    on_turn_on:
      - mhz19.abc_enable: mhz19sensor
    on_turn_off:
      - mhz19.abc_disable: mhz19sensor

uart:
 rx_pin: D4
 tx_pin: D3
 baud_rate: 9600

sensor:
 - platform: mhz19
   id: mhz19sensor
   co2:
    name: "MH-Z19 CO2 Keuken"
   temperature:
    name: "MH-Z19 Temperatuur Keuken"
   update_interval: 60s
   automatic_baseline_calibration: false

```
but it seems I'm not able to get it up and running:

```
INFO Connecting to co2keuken.local:6053 (192.168.1.186)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.186: [Errno 111] Connection refused). Re-Trying in 11 seconds
INFO Connecting to co2keuken.local:6053 (192.168.1.186)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.186: [Errno 111] Connection refused). Re-Trying in 17 seconds
INFO Connecting to co2keuken.local:6053 (192.168.1.186)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.186: [Errno 111] Connection refused). Re-Trying in 25 seconds
INFO Connecting to co2keuken.local:6053 (192.168.1.186)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.186: [Errno 111] Connection refused). Re-Trying in 30 seconds
INFO Connecting to co2keuken.local:6053 (192.168.1.186)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.186: [Errno 111] Connection refused). Re-Trying in 30 seconds

```

Looks like the api isn't accepting new connections. Will give the dev branch a try, see what happens.

kooskaspers, 2019-08-05:
dev branch works fine:
`[D][api.connection:576]: Client 'Home Assistant 0.96.5 (192.168.1.60)' connected successfully!`

I'll give it another try.

kooskaspers, 2019-08-05:
Got it working this time:
[image]

Will test upcoming day!

kooskaspers, 2019-08-09:
I've put both of my sensors outdoors for a few days, with ABC on. Disabled the ABC feature upon moving them back inside.  I have to admit it's working pretty good now:

[image]

Spirituss, 2019-08-15:
Have tested zero_calibration after one hour on open air and them ABC disabled - not good, the sensor shifts up the values more than 300 ppm in a couple of days. Seems that ABC to be disabled only after many hours on the open air, like one or two days at least. I will repeat test with such conditions and return back with results.

Anonym-tsk, 2019-08-15:
@Spirituss have you tested this feature with tasmota or espurna?
I'm not sure, but maybe it's problem with sensor, not firmware?

kooskaspers, 2019-08-17:
Still going strong:
[image]
Did you already file a PR for this branch @Anonym-tsk?

Anonym-tsk, 2019-08-19:
@kooskaspers i don't see autocalibration on this screenshot. What's wrong?

kooskaspers, 2019-08-19:
@Anonym-tsk I already performed ABC and turned it off now. The screenshot just proves that ABC is not kicking in anymore, which is the desired behavior, so I'm happy with your commits!

Spirituss, 2019-08-22:
I have been testing the sensor for some days and can approve that it works.