esphome/feature-requests #352: Support for Tuya Dimmers (Tuya TX / RX)

folkg, 2019-08-01:
**Describe the problem you have/What new integration you would like**

Implementation of the Tuya Dimmers functionality in ESPHome that is present in Tasmota firmware.

**Please describe your use case for this integration and alternatives you've tried:**

I have some Tuya dimmers at home that seem to use a secondary micro controller connected to the GPIO pins of the ESP. Tasmota implements this, but it would be great if they could be migrated to ESPHome.

**Additional context**

I have found some of the resources from the Tasmota docs that look to be applicable:

https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer

https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_16_tuyadimmer.ino

brandond, 2019-08-02:
These switch plates appear to use the same serial protocol as the Tuya PIR and door sensors that I have support for in my custom component.

https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_16_tuyadimmer.ino#L65
https://github.com/brandond/esphome-tuya_pir/blob/master/sb1_uart.h#L24

tortfeaser, 2019-08-05:
I'm not sure the link above to xrdv_16_tuyadimmer.ino is about the same tuya dimmer that Tasmota now supports and mentioned in the template at https://blakadder.github.io/templates/qs-wifi_D01_dimmer.html

If not, it would be great if it were supported. They're the cheapest (mosfet) wifi dimming device going, seemingly solid hardware.

arfrater, 2019-08-18:
I would like to request support for the Fan / Light Switch as discussed in this thread
https://github.com/arendst/Sonoff-Tasmota/issues/5650

[image]

brandond, 2019-08-19:
Anyone have a link to a North America version of these devices available for purchase? I'd pick one up and take a shot at it.

folkg, 2019-08-19:
Mine was from Costco in Canada, I'm not sure if there is an American equivalent.

https://www.costco.ca/CE-Smart-Home-Wi-Fi-Smart-Dimmer-Light-Switch%2c-2-pack.product.100417574.html

brandond, 2019-08-23:
I now have one of these: https://www.amazon.com/gp/product/B07PLCMR71/

They use a newer version of the Tuya MCU serial protocol than my custom component for battery-powered PIR and door sensors speak, but it's close enough that I think I can modify it pretty easily. I updated the protocol documentation here:
https://github.com/brandond/esphome-tuya_pir/blob/master/README.md#data-point-messages

I have a couple captures from my logic analyzer, and they match up pretty well with what the Tasmota code appears to be sending. I'll take a shot at putting together a component soon.

pr0ssi, 2019-09-07:
I have recently picked up a couple of the non-dimmer switches which are also Tuya. With these flashed with Tasmota the only config that works is the tuya-dimmer, though all buttons do the same thing. I've just flashed ESPHome, but not sure if these can't be switched without additional code due to these using the Tuya MCU serial protocol as well? A few photos to show the device in question
[image]
[image]
[image]

kvvoff, 2019-09-07:
@brandond
I admire your work done with the motion sensor!  This is amazing.  I would like to try to flash according to your instructions if I had a Tuya motion sensor.  Also I have 2 Tuya dimmers, one touch and one from Zemismart erc310. 
I am really looking forward to your dimmer implementation in ESPHome!

brandond, 2019-09-08:
FWIW, the dimmers use multiple Data Points to control the unit - a float data point for the dimmer and a bool for on/off. Since Tuya has six total primitive types (bool, int, float, string, enum, bytearray) I'll probably end up with components that map each of these to existing sensors or outputs. You'd just need to configure the ID number for each component, since each manufacturer is free to make up their own IDs and then map them to a control in the Tuya UI.

kvvoff, 2019-09-13:
@brandond I am ready to test my dimmers when you have a component for the test.

placidorevilla, 2019-09-18:
@brandond I wrote a prototype component for Tuya Dimmer a few months back, and I've been using it ever since: https://gist.github.com/placidorevilla/87f47c72b24021ff0c9eff622a521c39
Here's an incomplete example of how to use it: https://gist.github.com/placidorevilla/457afa39f801032e0af27c6a4ae89390

I only ran into one issue once, where the lights started flashing and I had to reset the controller. I haven't been able to reproduce it, but I'm assuming it might be some sort of desync of the protocol.

Feel free to do whatever you want with the code, since I haven't had time to propose for inclusion in the core esphome (and I don't think I will have a lot in the future either).

brandond, 2019-09-19:
Yeah I've just been short on time myself. Busy at work, plus a kitchen remodel and cyclocross season.

ssieb, 2019-10-08:
I've implemented this, so if anyone would like to ***carefully*** test it, that would be appreciated.  I only have one model to test with.  The reason I emphasize carefully is because some of these don't have accessible serial ports.  I bricked two of them while developing this.  One of them I ripped open so I could use the serial port, but that still required soldering directly to the esp module.

kvvoff, 2019-10-09:
@placidorevilla I tried your code with a dimmer Zemismart erc310. At the moment, this does not work. I can only turn it off, it works. The status also does not change if I turn it on manually

`[14:48:56][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x0F!=0x0E
[14:48:56][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x18!=0x17
[14:48:56][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x7E!=0x7D
[14:48:56][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x0F!=0x0E
[14:49:00][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01`

ssieb, 2019-10-09:
I'll put back the option to log the incoming messages to see what's happening.  What devices did it detect?

placidorevilla, 2019-10-09:
@kvvoff There might be a bug in the checksum calculation code. I only have one device to test this in, so can you insert the following line between lines 181 and 182 of tuyadimmer_lightoutput.h? This will dump the packet in Hex and I might be able to figure out what's going on here.

`this->dump_packet_();`

placidorevilla, 2019-10-09:
BTW, this is the device I have: https://www.aliexpress.com/item/32980110228.html
I had to solder tiny wires to the ESP module as well to flash it.

kvvoff, 2019-10-09:
Does your code work with this device?  I was thinking about buying this dimmer.  
My dimmers:
https://s.click.aliexpress.com/e/bvyXfrwS
https://s.click.aliexpress.com/e/bzbfhCgw

kvvoff, 2019-10-10:
@placidorevilla When I add this line to tuyadimmer_lightoutput.h,
```
      if (checksum != byte) {
        ESP_LOGW(TAG, "Checksum doesn't match: 0x%02X!=0x%02X", byte, checksum);
        this->dump_packet_();
        return false;
      }
    }
```
 compilation fails.
`
Compiling /data/erc310/.pioenvs/erc310/src/main.cpp.o
In file included from src/main.cpp:13:0:
src/tuyadimmer_lightoutput.h: In member function 'esphome::optional<bool> TuyaDimmerLightOutput::check_byte_() const':
src/tuyadimmer_lightoutput.h:182:28: error: passing 'const TuyaDimmerLightOutput' as 'this' argument of 'void TuyaDimmerLightOutput::dump_packet_()' discards qualifiers [-fpermissive]
this->dump_packet_();
^
*** [/data/erc310/.pioenvs/erc310/src/main.cpp.o] Error 1
========================== [ERROR] Took 3.27 seconds ==========================
`

 What am I doing wrong?

placidorevilla, 2019-10-10:
@kvvoff Oh, please, change the definition of dump_packet_ in line 189 to:

`void dump_packet_() const {`

That method should have been const since the beginning.

kvvoff, 2019-10-11:
@placidorevilla I did this, here is the output of my log:

`[12:34:33][D][debug:023]: ESPHome version 1.13.6
[12:34:33][D][debug:025]: Free Heap Size: 40280 bytes
[12:34:33][D][debug:053]: Flash Chip: Size=1024kB Speed=40MHz Mode=DOUT
[12:34:33][D][debug:190]: Chip ID: 0x007491CD
[12:34:33][D][debug:191]: SDK Version: 2.2.1(cfd48f3)
[12:34:33][D][debug:192]: Core Version: 2_4_2
[12:34:33][D][debug:193]: Boot Version=4 Mode=1
[12:34:33][D][debug:194]: CPU Frequency: 80
[12:34:33][D][debug:195]: Flash Chip ID=0x001440C8
[12:34:33][D][debug:196]: Reset Reason: Software/System restart
[12:34:33][D][debug:197]: Reset Info: Fatal exception:0 flag:4 (SOFT_RESTART) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[12:34:35][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:34:35][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:34:45][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:34:45][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:34:54][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x16!=0x15
[12:34:54][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 07 00 08 03 02 00 04 00 00 00 FE
[12:34:54][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x10!=0x0F
[12:34:54][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 07 00 05 01 01 00 01 01
[12:34:54][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x7E!=0x7D
[12:34:54][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 07 00 08 69 02 00 04 00 00 00 00
[12:34:55][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:34:55][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:34:56][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x0F!=0x0E
[12:34:56][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 07 00 05 01 01 00 01 00
[12:34:56][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x18!=0x17
[12:34:56][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 07 00 08 03 02 00 04 00 00 00 00
[12:34:56][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x7E!=0x7D
[12:34:56][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 07 00 08 69 02 00 04 00 00 00 00
[12:35:05][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:35:05][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:35:15][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:35:15][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:35:25][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:35:25][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:35:35][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:35:35][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:35:45][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:35:45][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:35:55][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:35:55][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:36:05][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:36:05][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:36:16][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:36:16][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:36:19][I][ota:046]: Boot seems successful, resetting boot loop counter.
[12:36:25][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:36:25][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:36:35][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:36:35][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01
[12:36:45][W][tuyadimmerlightoutput:181]: Checksum doesn't match: 0x02!=0x01
[12:36:45][D][tuyadimmerlightoutput:195]: Packet:  55 AA 01 00 00 01 01`

Once I turned it on with the button on the device, then turned it off with the button on the device.

placidorevilla, 2019-10-11:
@kvvoff Oh, I think I know what's going on. I misinterpreted something in the specification, the version byte needs to be included in the checksum calculation (but in my device, version is 0, so it doesn't affect the checksum, while in yours, version is 1, and that's why the checksum is off by 1).
Can you change the line 177 for:

`      for (int i = 2; i < index; ++i) {`

That should do it.

brandond, 2019-10-11:
I am working on a more generic version of this that will let you map any Tuya DPiD to a sensor or output. It's not ready for release yet though :/

ssieb, 2019-10-11:
I have a linked pull request to handle dimmers.  Adding a sensor or output would be trivial.

kvvoff, 2019-10-11:
@brandond When you have a test version, let me know.  I can test it on my dimmers

kvvoff, 2019-10-11:
@placidorevilla Thanks, I will check it after the weekend, now there is no way

kvvoff, 2019-10-13:
@ssieb I want to check your implementation, but I don’t have the opportunity now to install your version of esphome. Can you make a custom_components for me to check?

ssieb, 2019-10-13:
You can download http://sieb.net/tuya.zip and extract it in the components folder of your esphome installation.  There should now be a tuya folder in your components.  I rendered my updated doc to http://www.sieb.net/components/light/tuyadimmer.html for instructions.

kvvoff, 2019-10-14:
@placidorevilla I changed line 177. Now there are no checksum errors in the log. At the moment, I can not turn on the dimmer or change the brightness. **I can only turn it off** if it is on. When I turn the dimmer on or off (not using esphome+ha, manually) the status changes in esphome+ha. But **only the on / off status changes**. The brightness status (slider) is not true.
`[12:54:41][D][debug:197]: Reset Info: Fatal exception:0 flag:4 (SOFT_RESTART) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[12:54:43][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:54:48][D][light:261]: 'Dimmer Light' Setting:
[12:54:48][D][light:270]:   State: ON
[12:54:48][D][tuyadimmerlightoutput:057]: Set output to 1.000000 (163)
[12:54:53][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:54:59][D][light:261]: 'Dimmer Light' Setting:
[12:54:59][D][light:274]:   Brightness: 46%
[12:54:59][D][tuyadimmerlightoutput:057]: Set output to 0.458824 (103)
[12:55:01][D][light:261]: 'Dimmer Light' Setting:
[12:55:01][D][light:274]:   Brightness: 16%
[12:55:01][D][tuyadimmerlightoutput:057]: Set output to 0.156863 (70)
[12:55:03][D][light:261]: 'Dimmer Light' Setting:
[12:55:03][D][light:274]:   Brightness: 65%
[12:55:03][D][tuyadimmerlightoutput:057]: Set output to 0.647059 (124)
[12:55:03][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:55:04][D][light:261]: 'Dimmer Light' Setting:
[12:55:04][D][light:274]:   Brightness: 89%
[12:55:04][D][tuyadimmerlightoutput:057]: Set output to 0.890196 (151)
[12:55:06][D][light:261]: 'Dimmer Light' Setting:
[12:55:06][D][light:274]:   Brightness: 47%
[12:55:06][D][tuyadimmerlightoutput:057]: Set output to 0.470588 (105)
[12:55:07][D][light:261]: 'Dimmer Light' Setting:
[12:55:07][D][light:274]:   Brightness: 24%
[12:55:07][D][tuyadimmerlightoutput:057]: Set output to 0.243137 (80)
[12:55:08][D][light:261]: 'Dimmer Light' Setting:
[12:55:08][D][light:274]:   Brightness: 7%
[12:55:09][D][tuyadimmerlightoutput:057]: Set output to 0.070588 (61)
[12:55:12][D][light:261]: 'Dimmer Light' Setting:
[12:55:12][D][light:274]:   Brightness: 34%
[12:55:12][D][tuyadimmerlightoutput:057]: Set output to 0.337255 (90)
[12:55:13][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:55:15][D][light:261]: 'Dimmer Light' Setting:
[12:55:15][D][light:274]:   Brightness: 84%
[12:55:15][D][tuyadimmerlightoutput:057]: Set output to 0.835294 (145)
[12:55:16][D][light:261]: 'Dimmer Light' Setting:
[12:55:16][D][light:274]:   Brightness: 96%
[12:55:16][D][tuyadimmerlightoutput:057]: Set output to 0.956863 (158)
[12:55:19][D][light:261]: 'Dimmer Light' Setting:
[12:55:19][D][light:274]:   Brightness: 30%
[12:55:19][D][tuyadimmerlightoutput:057]: Set output to 0.298039 (86)
[12:55:20][D][light:261]: 'Dimmer Light' Setting:
[12:55:20][D][light:274]:   Brightness: 66%
[12:55:20][D][tuyadimmerlightoutput:057]: Set output to 0.658824 (125)
[12:55:23][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:55:24][D][light:261]: 'Dimmer Light' Setting:
[12:55:24][D][light:270]:   State: OFF
[12:55:24][D][tuyadimmerlightoutput:210]: Power State: 0x00
[12:55:26][D][light:261]: 'Dimmer Light' Setting:
[12:55:26][D][light:270]:   State: ON
[12:55:26][D][tuyadimmerlightoutput:057]: Set output to 0.658824 (125)
[12:55:28][D][light:261]: 'Dimmer Light' Setting:
[12:55:28][D][light:270]:   State: OFF
[12:55:28][D][tuyadimmerlightoutput:210]: Power State: 0x00
[12:55:33][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:55:43][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:55:53][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:56:03][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:56:13][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:56:23][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:56:27][I][ota:046]: Boot seems successful, resetting boot loop counter.
[12:56:33][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:56:43][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:56:53][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:57:03][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:57:13][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:57:23][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:57:33][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[12:57:43][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[... 146 lines cut ...]
[13:12:56][W][tuyadimmerlightoutput:217]: Unknown dpID: 0x03
[13:12:56][W][tuyadimmerlightoutput:217]: Unknown dpID: 0x69
[13:12:56][D][tuyadimmerlightoutput:210]: Power State: 0x00
[13:13:04][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[13:13:14][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[13:13:24][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[13:13:33][W][tuyadimmerlightoutput:217]: Unknown dpID: 0x03
[13:13:33][D][tuyadimmerlightoutput:210]: Power State: 0x01
[13:13:33][D][light:261]: 'Dimmer Light' Setting:
[13:13:33][D][light:270]:   State: ON
[13:13:33][D][tuyadimmerlightoutput:057]: Set output to 0.533333 (112)
[13:13:33][W][tuyadimmerlightoutput:217]: Unknown dpID: 0x69
[13:13:34][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[13:13:37][D][tuyadimmerlightoutput:210]: Power State: 0x01
[13:13:37][W][tuyadimmerlightoutput:217]: Unknown dpID: 0x03
[13:13:37][W][tuyadimmerlightoutput:217]: Unknown dpID: 0x69
[13:13:42][D][tuyadimmerlightoutput:210]: Power State: 0x01
[13:13:42][W][tuyadimmerlightoutput:217]: Unknown dpID: 0x03
[13:13:42][W][tuyadimmerlightoutput:217]: Unknown dpID: 0x69
[13:13:44][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[13:13:54][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[13:14:04][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[13:14:14][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[13:14:24][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)
[13:14:34][D][tuyadimmerlightoutput:201]: MCU Heartbeat (0x01)`

kvvoff, 2019-10-14:
@ssieb I checked your code. My dimmer switches on and off from esphome, and the status also changes if I change manually. 

But there is one problem. When I set the brightness **from 19 to 100** everything is fine. But when changing the brightness **from 0 to 19**, the dimmer shines at **full power.** It seems that **0 = 100%** brightness, because when I put the slider at 0 it shines much brighter than when the slider is at 100.
Almost the same problem was on the Tasmot firmware, but when I wrote about this, no one paid attention.

ssieb, 2019-10-14:
Set the min value to 20.

kvvoff, 2019-10-14:
@ssieb error compilation if I set min_value 20

`light:
  - platform: "tuya"
    name: "dim1"
    dimmer: 3
    switch: 1
    min_value: 20
    max_value: 255
    gamma_correct: 1.0
    default_transition_length: 0s`

`Linking .pioenvs/erc310/firmware.elf
.pioenvs/erc310/src/main.cpp.o:(.text.setup+0x68): undefined reference to `esphome::tuya::TuyaLight::set_min_value(unsigned int)'
.pioenvs/erc310/src/main.cpp.o:(.text.setup+0x6c): undefined reference to `esphome::tuya::TuyaLight::set_max_value(unsigned int)'
.pioenvs/erc310/src/main.cpp.o: In function `setup':
main.cpp:(.text.setup+0x3cb): undefined reference to `esphome::tuya::TuyaLight::set_min_value(unsigned int)'
main.cpp:(.text.setup+0x3d6): undefined reference to `esphome::tuya::TuyaLight::set_max_value(unsigned int)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/erc310/firmware.elf] Error 1
============================= [FAILED] Took 5.74 seconds =============================`

placidorevilla, 2019-10-14:
@kvvoff in my component the dimmer id defaults to 2, while you're using 3 with @ssieb code. This is unfortunately one of the most annoying things about these dimmers, since, as far as I know there's no good way to figure out programmatically what these ids should be.
With my code, when you instantiate the component you can pass the ids there, just instantiate them like this:
`
auto tuyadimmer = new TuyaDimmerLightOutput(id(uart1), 1, 3);
`
You can also change the min and max power there, with the 4th and 5th parameters (they default to 0.21 and 0.64 because that's what my dimmer will do, again, no way to figure this out except by experimentation).

That said, your log has references to dpid 0x69 as well, which I don't know what it could be.

ssieb, 2019-10-14:
@kvvoff I apologize for not testing those options.  I'm a little confused about why how it compiled at all like that.  I've pushed a fix and updated the zip file.
Can you please paste the config output that lists the found devices.

kvvoff, 2019-10-15:
@ssieb We are here to test, so don’t worry about it)
I set `min_value: 20` with your new tuya.zip, and the compilation was successful. Now my dimmer works perfectly! Thank you)
This is my zemismart erc310 dimmer, when there is time I will test my other dimmer Norklmes

kvvoff, 2019-10-15:
@placidorevilla Empirically, I found out that for my erc310 dimmer `static const float DEFAULT_MIN_POWER = 0.08`
I also set `static const uint8_t DEFAULT_DIMMER_ID = 3`

Now my dimmer works perfectly with your code. Thank you :)

kvvoff, 2019-10-15:
I also found interesting features of both versions of the code @ssieb and @placidorevilla 
At @ssieb, when the dimmer is turned off and I turn it on from esphome and HA, the brightness is set to 100%, and it does not matter what brightness it was before. That is, your code does not remember the brightness, and when you turn on the command from esphome+HA, it is at maximum. The @placidorevilla code does not have this, and when turned on, the brightness that was before that is turned on.

Also in code @ssieb, when I turn on the dimmer via a physical button, it also does not remember the brightness and sets about ~40%. I do not know why. Code @placidorevilla  does not have this, and the brightness is turned on as it was before

ssieb, 2019-10-16:
On mine, the dimmer keeps its value.  I can toggle the "switch" and it will go off and on but still have the same brightness when on.  The physical switch has the same action.  If you turn on VERBOSE logging, mine will show you the messages it's getting from the device.  It would be interesting to see if it shows the value changing.  I'm also curious if you get anything from that unknown value as well.

kvvoff, 2019-10-18:
@ssieb I turned on the switch at HA, set the brightness 26%, turned off the switch at HA. Then I turned on the switch and the brightness became 100%, although it was 26%
Then I set the brightness to 45, turned it off and on with the button, the brightness returned to its previous state

`[08:55:16][C][wifi:372]: WiFi:

[08:55:16][C][wifi:254]:   SSID: [redacted]
[08:55:16][C][wifi:255]:   IP Address: 192.168.0.22
[08:55:16][C][wifi:257]:   BSSID: [redacted]
[08:55:16][C][wifi:258]:   Hostname: 'erc310'
[08:55:16][C][wifi:262]:   Signal strength: -49 dB ▂▄▆█
[08:55:16][C][wifi:263]:   Channel: 1
[08:55:16][C][wifi:264]:   Subnet: 255.255.255.0
[08:55:16][C][wifi:265]:   Gateway: 192.168.0.1
[08:55:16][C][wifi:266]:   DNS1: 192.168.0.1
[08:55:16][C][wifi:267]:   DNS2: 0.0.0.0
[08:55:16][C][uart:128]: UART Bus:
[08:55:16][C][uart:130]:   TX Pin: GPIO1
[08:55:16][C][uart:133]:   RX Pin: GPIO3
[08:55:16][C][uart:135]:   Baud Rate: 9600 baud
[08:55:16][C][uart:137]:   Using hardware serial interface.
[08:55:16][C][tuya:057]: Tuya:
[08:55:16][C][tuya:062]:   1: switch
[08:55:16][C][tuya:064]:   3: int value
[08:55:16][C][tuya:064]:   105: int value
[08:55:16][C][logger:137]: Logger:
[08:55:16][C][logger:138]:   Level: VERBOSE
[08:55:16][C][logger:139]:   Log Baud Rate: 0
[08:55:16][C][logger:140]:   Hardware UART: UART0
[08:55:16][C][light:174]: Light 'dim1'
[08:55:16][C][light:176]:   Default Transition Length: 0.0s
[08:55:16][C][light:177]:   Gamma Correct: 1.00
[08:55:16][C][tuya.light:023]: Tuya Dimmer:
[08:55:16][C][tuya.light:025]:    dimmer is dpid 3
[08:55:16][C][tuya.light:027]:    switch is dpid 1
[08:55:16][C][ota:029]: Over-The-Air Updates:
[08:55:16][C][ota:030]:   Address: erc310.local:8266
[08:55:16][C][ota:032]:   Using Password.
[08:55:16][C][api:103]: API Server:
[08:55:16][C][api:104]:   Address: erc310.local:6053
[08:55:16][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:55:17][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:55:18][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:55:19][V][api:542]: Hello from client: 'Home Assistant 0.100.1 (192.168.0.16)'
[08:55:19][D][api:573]: Client 'Home Assistant 0.100.1 (192.168.0.16)' connected successfully!
[08:55:19][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:56:03][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:56:07][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:56:07][D][light:261]: 'dim1' Setting:
[08:56:07][D][light:270]:   State: ON
[08:56:07][V][tuya:139]: command 7 version 1 count 8 data [03 02 00 04 00 00 00 ff ] checksum 23 23
[08:56:07][V][tuya:192]: dp 3 update to 255
[08:56:07][V][tuya.light:041]: dp_update(3, 255)
[08:56:07][V][tuya.light:049]: dimmer update to 255
[08:56:07][D][light:261]: 'dim1' Setting:
[08:56:07][D][light:274]:   Brightness: 100%
[08:56:07][V][tuya:139]: command 7 version 1 count 5 data [67 01 00 01 00 ] checksum 117 117
[08:56:07][V][tuya:192]: dp 103 update to 0
[08:56:07][V][tuya:139]: command 7 version 1 count 5 data [67 01 00 01 00 ] checksum 117 117
[08:56:11][V][tuya:192]: dp 103 update to 0
[08:56:11][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:56:11][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:56:11][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:56:11][D][light:261]: 'dim1' Setting:
[08:56:11][D][light:274]:   Brightness: 26%
[08:56:11][V][tuya:139]: command 7 version 1 count 8 data [03 02 00 04 00 00 00 43 ] checksum 91 91
[08:56:11][V][tuya:192]: dp 3 update to 67
[08:56:11][V][tuya.light:041]: dp_update(3, 67)
[08:56:11][V][tuya.light:049]: dimmer update to 67
[08:56:11][D][light:261]: 'dim1' Setting:
[08:56:11][D][light:274]:   Brightness: 26%
[... 71 lines cut ...]
[08:56:35][V][tuya:192]: dp 3 update to 0
[08:56:35][V][tuya.light:041]: dp_update(3, 0)
[08:56:35][V][tuya.light:049]: dimmer update to 0
[08:56:35][D][light:261]: 'dim1' Setting:
[08:56:35][D][light:274]:   Brightness: 100%
[08:56:35][V][tuya:139]: command 7 version 1 count 8 data [69 02 00 04 00 00 00 00 ] checksum 126 126
[08:56:35][V][tuya:192]: dp 105 update to 0
[08:56:35][V][tuya:139]: command 7 version 1 count 5 data [01 01 00 01 00 ] checksum 15 15
[08:56:35][V][tuya:192]: dp 1 update to 0
[08:56:36][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2
[08:56:37][V][tuya:139]: command 7 version 1 count 8 data [03 02 00 04 00 00 00 72 ] checksum 138 138
[08:56:37][V][tuya:192]: dp 3 update to 114
[08:56:37][V][tuya.light:041]: dp_update(3, 114)
[08:56:37][V][tuya.light:049]: dimmer update to 114
[08:56:37][D][light:261]: 'dim1' Setting:
[08:56:37][D][light:274]:   Brightness: 45%
[08:56:37][V][tuya:139]: command 7 version 1 count 5 data [01 01 00 01 01 ] checksum 16 16
[08:56:37][V][tuya:192]: dp 1 update to 1
[08:56:37][V][tuya.light:041]: dp_update(1, 1)
[08:56:37][V][tuya.light:054]: switch update to 1
[08:56:37][D][light:261]: 'dim1' Setting:
[08:56:37][D][light:270]:   State: ON
[08:56:37][V][tuya:139]: command 7 version 1 count 8 data [69 02 00 04 00 00 00 00 ] checksum 126 126
[08:56:37][V][tuya:192]: dp 105 update to 0
[08:56:38][V][tuya:139]: command 0 version 1 count 1 data [01 ] checksum 2 2`

ssieb, 2019-10-19:
I've updated the zip file and the documentation.  The dimmer and switch parameters are now dimmer_datapoint and switch_datapoint.  I've also added logging (with VERBOSE) to see what data values are being sent to the MCU as well.

closed 2019-10-19 by OttoWinter

tortfeaser, 2019-10-25:
Hi there. Thanks for working to get this integrated. I've tried the tuya: component in both the dev and beta branches and am not getting any response. Here's the log output:

```
[09:40:11][I][app:100]: esphome version 1.14.0b4 compiled on Oct 26 2019, 09:37:26
[09:40:11][C][wifi:409]: WiFi:
[09:40:11][C][wifi:277]:   SSID: 'FISHnet'[redacted]
[09:40:11][C][wifi:278]:   IP Address: 192.168.1.180
[09:40:11][C][wifi:280]:   BSSID: 2C:30:33:4D:B3:B3[redacted]
[09:40:11][C][wifi:281]:   Hostname: 'wifi_dimmer_2'
[09:40:11][C][wifi:285]:   Signal strength: -65 dB ▂▄▆█
[09:40:11][V][wifi:287]:   Priority: 0.0
[09:40:11][C][wifi:289]:   Channel: 6
[09:40:11][C][wifi:290]:   Subnet: 255.255.255.0
[09:40:11][C][wifi:291]:   Gateway: 192.168.1.1
[09:40:11][C][wifi:292]:   DNS1: 192.168.1.1
[09:40:11][C][wifi:293]:   DNS2: (IP unset)
[09:40:11][C][uart:134]: UART Bus:
[09:40:11][C][uart:136]:   TX Pin: GPIO15
[09:40:11][C][uart:139]:   RX Pin: GPIO13
[09:40:11][C][uart:141]:   Baud Rate: 9600 baud
[09:40:11][C][uart:142]:   Stop bits: 1
[09:40:11][C][uart:144]:   Using hardware serial interface.
[09:40:11][C][tuya:024]: Tuya:
[09:40:11][C][logger:175]: Logger:
[09:40:11][C][logger:176]:   Level: VERBOSE
[09:40:11][C][logger:177]:   Log Baud Rate: 0
[09:40:11][C][logger:178]:   Hardware UART: UART0
[09:40:11][C][ota:029]: Over-The-Air Updates:
[09:40:11][C][ota:030]:   Address: wifi_dimmer_2.local:8266
[09:40:11][C][ota:032]:   Using Password.
[09:40:11][C][api:095]: API Server:
[09:40:11][C][api:096]:   Address: wifi_dimmer_2.local:6053
```
No output from the tuya: saying what the datapoint values are. I get the same result with TX/RX on GPIO1/3 as in https://blakadder.github.io/templates/qs-wifi_D01_dimmer.html. If I plug in the datapoint values from http://www.sieb.net/components/light/tuyadimmer.html and create the light: component like on that page, I get no response from the dimmer or switch.

Any ideas?

ssieb, 2019-10-26:
If the Tuya: section doesn't list the datapoints, then it's not going to work.  You need to figure out which pins have the serial connection.  Are you able to non-destructively open the device?  What is the brand and model?

tortfeaser, 2019-10-26:
I've got the device working with Tasmota via this template: https://blakadder.github.io/templates/qs-wifi_D01_dimmer.html My device is (exactly) the same as on that page (including the box and internals). It came in a box labelled 'QS-WIFI-D01-TRIAC'. 

It uses TX/RX on GPIO1/3. 

I've tried uploading the esphome-compiled firmware via the Tasmota interface, including after uploading the smallest Tasmota .bin (tasmota-minimal) and erasing flash.

Mystery.

OttoWinter, 2019-10-26:
The page you just posted has the answer: "The serial communication at 9600 baud is simple, just one command with changing brightness xx values: FF 55 xx 05 DC 0A". That is not the Tuya protocol implemented here, it is a different one (probably custom made for this chip). This integration only works for the standard-ish tuya protocol.

Also, for anybody having issues with the tuya integration, please create a full issue report (possibly with a link pointing to here for reference).