esphome/feature-requests #356: Nextion: call "printf" for text sensors

romquenin, 2019-08-02:
**Describe the problem you have/What new integration you would like**

Hello i have succesfully  linked buttons and sensors with "numeric" state from home assistant to my display but unfortunately "text" sensors wont work.

I'm completely stuck this time and and would to know if you could me with this issue.
Here is the code i have for now:

`          it.set_component_text_printf("date","%",id(date_jour).state);`

I have tried to add "d", and "s" after the percent but "d" is retuning nothing an "s" makes the display become crazy.

Any help would be awsome.

Thank you.

Love your work and ESPHome,

romquenin, 2019-08-02:
```
sensor:
  - platform: homeassistant   # Forecast precipitation
    id: uptime_hours
    entity_id: sensor.uptime     

text_sensor:
  - platform: homeassistant
    id: date_jour
    entity_id: weather.dark_sky
  - platform: version
    name: ESPHome Version
    id: version_test

            
binary_sensor:
  - platform: nextion
    page_id: 0
    component_id: 14
    id: reboot
    on_press:
      then:
        - homeassistant.service:
            service: script.hdmi_4

display:
  - platform: nextion
    id: teplomer
    update_interval: 5s
    lambda: |-
          it.set_component_text_printf("ligne","EN LIGNE DEPUIS %.2f MINUTES",id(uptime_hours).state);
          it.set_component_text_printf("date","%s",id(date_jour).state);
```

romquenin, 2019-08-02:
[image]

romquenin, 2019-08-03:
Here is a link to my video tutorial on how to build a ESPHome LCD Weather Station:

https://youtu.be/lCuXGcnOQ7g

I can't wait to understand how to integrate majors Nextion editor options with ESPHome (gauge, progress bar,..).

glmnet, 2019-08-03:
You need to add `.c_str()` after `state`

romquenin, 2019-08-04:
Thank you so much @glmnet ! It works now. 

Do you have any idea on how to set the gauge and the progress bar in the esphome configuration file ?

That would be the cherry on the cake !

Thanks.

OttoWinter, 2019-10-13:
Closing as OP's initial issue is fixed.

Don't know about gauge and progress bar, but I would suggest you look at the Nextion Instruction Set manual (google). It says exactly how all these commands are sent.

closed 2019-10-13 by OttoWinter