esphome/feature-requests #357: Web based UI and/or Control option

charudatt, 2019-08-03:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

A web based UI for configuration of the Hardware  and also Web based trigger , just like TASMOTA has. In which case the Hardware also becomes standalone and operated via Web UI.

Please include a WiFi manager so that you can select a HotSpot via Web UI

glmnet, 2019-08-04:
Hardware configuration over web ui isn’t going to happen:
ESPHome is a tool which reads in YAML configuration files (just like Home Assistant) and creates custom firmwares. So running esphome “full stack” inside esphome “runtime” sounds exciting but I believe ESPs aren’t powerful enough yet. 

The WiFi config is coming see https://github.com/esphome/feature-requests/issues/86

As for the control from the web I believe a ...

````yaml
switch:
  - platform: web
     ...
````

Might be useful..

OttoWinter, 2019-10-13:
Fully agree with @glmnet, not going to happen. If you want a GUI to configure devices, use tasmota. If you want the power of being able to set every little detail yourself and even add custom code you can look at esphome.

closed 2019-10-13 by OttoWinter