esphome/feature-requests #358: Support for Sonoff RF Bridge

soppilif, 2019-08-03:
**Describe the problem you have/What new integration you would like**

Support for Sonoff RF Bridge
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

rradar, 2019-08-03:
> **Note**
> For the Sonoff RF Bridge you can use this hack created by the Github user wildwiz. Then use this configuration for the remote receiver/transmitter hubs:
> 
> ```
> remote_receiver:
>   pin: 4
>   dump: all
> 
> remote_transmitter:
>   pin: 5
>   carrier_duty_percent: 100%
> ```

https://esphome.io/components/remote_receiver.html?highlight=Sonoff%20RF%20Bridge#id1

vladiemier, 2019-09-01:
I've used this custom yaml from pinkywafer and changed it to my needs.

https://github.com/pinkywafer/sonoff_rf_bridge-esphome

The mod mentioned above needs to be carried out but i've had this running for almost a month and it's been great!

deslizador27, 2019-11-09:
Esphome es genial. Y cada día amplian mas su compatibilidad con mas sensores

closed 2022-07-06 by nagyrobi