esphome/feature-requests #359: Several intervals

HFsi, 2019-08-07:
**Describe the problem you have/What new integration you would like**

I need to define more than one interval, to do certain operations
**Please describe your use case for this integration and alternatives you've tried:**

I have for example an interval that blinks a led in N seconds depending on the state of one sensor, but I need another interval for switching a rele under certain periodic conditions
**Additional context**

glmnet, 2019-08-07:
Hi,

You can have several intervals:

````yaml
# Example configuration entry
interval:
  - interval: 1min
    then:
      - switch.toggle: relay_1
  - interval: 42s
    then:
      - light.toggle: light_1
````

closed 2019-08-07 by glmnet

HFsi, 2019-08-07:
Wow, thanks man!
Maybe this could be stated in docs, for clarification

glmnet, 2019-08-07:
It is clear enough, because of the dash: `-` before `interval: 1min` that means you can have many items.
Learning yaml takes a bit of time. You'll like it.