esphome/feature-requests #360: Support mdi icon for binary sensor

tomlut, 2019-08-08:
**Describe the problem you have/What new integration you would like**

Add support for mdi icons to the binary_sensor component. This is available for sensor and switch components.

**Please describe your use case for this integration and alternatives you've tried:**

No actual problem. This can be overcome with customization in Home Assistant, it would just be nice not to have to.

**Additional context**

brandond, 2019-08-09:
You can set `device_class` on binary sensors, which does control the icon in HA. Does that get you close to what you want?

glmnet, 2019-08-09:
close may be... but if it is supported on sensors and switches as @tomlut says... then why not on binary_sensors
this feature will add consistency

tomlut, 2019-08-10:
In the case where a class is not available it would be nice to be able to set the icon in the ESPhome config rather than customize (or Lovelace) in Home Assistant.

This way the ESPhome device is discovered with nothing left to set up.

OttoWinter, 2019-08-11:
I don't have anything to object to this. If someone implements it, I would be happy to merge - but beware it's quite a bit of work, stuff that needs to be updated would include:

 - ESPHome itself (obviously)
 - aioesphomeapi
 - Home Assistant esphome integration

Probably also good to update MQTT integration, but optional:

 - Add icon to HA mqtt binary sensor
 - Same for esphome mqtt binary sensor side.

closed 2021-10-10 by OttoWinter