esphome/feature-requests #361: Publish Cover Current Operation

gitlemat, 2019-08-09:
**Describe the problem you have/What new integration you would like**

Currently ESPHome publishes the position/state of the coers.
It would be very interesting to have the current_operation also published.

**Please describe your use case for this integration and alternatives you've tried:**

We can actually obtain this same behavior with a couple of config changes (adding the publishes), but it may create later issues if we have any braking change.

**Additional context**

glmnet, 2019-08-09:
Silly question may be, why we need this?

gitlemat, 2019-08-09:
In home-assistant we have ways to show if the covers are moving or not, and for that we need to notify the current_operation.

TBH I have not checked if this is done with the native API since I'm only using MQTT for compatibility reasons.

glmnet, 2019-08-09:
I'm using the covers with the api.
I've created this issue in Home Assistant may be you can have some useful information from there: https://github.com/home-assistant/home-assistant/issues/25738
I did not log what the ESPHome tells HA, but I did log what the HA tells to Home Kit, and I see it saying: "Openning"

OttoWinter, 2019-10-13:
> It would be very interesting to have the current_operation also published.

Since 1.13 that is the case. See `current_operation` on the Cover object.

closed 2019-10-13 by OttoWinter