esphome/feature-requests #363: Add a "start_at" attribute to fastled and neopixelbus

mplinuxgeek, 2019-08-11:
**Describe the problem you have/What new integration you would like**
I plan on printing [these](https://www.thingiverse.com/thing:3372353) and with the way ESPHome currently works I would need a "controller" in each segment, I suggest adding a "start_at" attribute, the idea being you could create a light as the first 9 LED's, then define a new light using LED's 10 through 18 and so on.
```
- platform: neopixelbus
    name: "RGB Light 1"
    type: GRB
    pin: GPIO23
    num_leds: 9
    start_at: 0
- platform: neopixelbus
    name: "RGB Light 2"
    type: GRB
    pin: GPIO23
    num_leds: 9
    start_at: 10
- platform: neopixelbus
    name: "RGB Light 3"
    type: GRB
    pin: GPIO23
    num_leds: 9
    start_at: 19
- platform: neopixelbus
    name: "RGB Light 4"
    type: GRB
    pin: GPIO23
    num_leds: 9
    start_at: 27
```

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

jchambers2012, 2019-08-12:
I think you are looking for Light Partition module.  It is what I use to sub-divide a LED strip:
https://esphome.io/components/light/partition.html

jchambers2012, 2019-08-13:
I didnt have access to my house when i last posted, here an example of my setup:
'''
light:
  - platform: neopixelbus
    variant: SK6812
    internal: true
    pin: D4
    num_leds: 231
    type: GRBW
    name: "Kitchen Lights"
    id: kitchen_lights
    method: ESP8266_UART1
    effects:
      - addressable_rainbow:
      - addressable_rainbow:
          name: "Rainbow Effect With Custom Values"
          speed: 200
          width: 300
  - platform: partition
    name: "Kitchen Cabinets Upper"
    id: kitchen_cabinets_upper
    segments:
      - id: kitchen_lights
        from: 0
        to: 127
    effects:
      - addressable_rainbow:
          width: 300
      - addressable_rainbow:
          name: "Rainbow Effect With Custom Values"
          speed: 100
          width: 300
  - platform: partition
    name: "Kitchen Cabinets Under"
    id: kitchen_cabinets_under
    segments:
      - id: kitchen_lights
        from: 128
        to: 200
    effects:
      - addressable_rainbow:
          width: 300
      - addressable_rainbow:
          name: "Rainbow Effect With Custom Values"
          speed: 50
          width: 300
'''

mplinuxgeek, 2019-08-14:
@jchambers2012 thank you, I don't know how I missed that module, will give it a go.

mplinuxgeek, 2019-08-14:
@jchambers2012 thanks again, I've had a play and it does exactly what I need... closing this feature request.

closed 2019-08-14 by mplinuxgeek