esphome/feature-requests #364: Rolling median filter

petersandersen, 2019-08-11:
**Describe the problem you have/What new integration you would like**
Provide the option to smooth sensor readings with a rolling median instead of the rolling average.

**Please describe your use case for this integration and alternatives you've tried:**

Some sensors sensors have large spikes in the output value that will distort rolling averages so I'd like to smooth them by taking the median over a window to avoid the impact of outliers / extreme values. E.g. every couple of hours I get a spike of 500 on my SDS011 particle counter, I don't want to filter out these values but a single reading shouldn't be able to move the average reading that much.

**Additional context**

Only need small window sizes, most sensors would work well with the middle number of the last 3 or 5 readings.
Also thanks heaps for your work on ESPHome, its really opened up a new world in how I use home assistant and tinker with electronics I never thought I'd be touching. Very impressed with how easy and clean its been to get things that just work setup.

glmnet, 2019-08-11:
Just to be sure. This calc?
 
In any given set, the arithmetic mean is the average, which is the sum total of the numbers divided by how many numbers there are. The median is the middle number of any given set when they are in order, least to greatest. If there is an even number of values, the median is the average of the middle two.

petersandersen, 2019-08-11:
Yep thats it exactly. Thanks for checking

hSoubry, 2019-08-16:
great idea. Would we a good filter for the ultrasonic sensor too.