esphome/feature-requests #365: Not a new feature, new device guide for Sonoff Mini?

tortfeaser, 2019-08-16:
**Describe the problem you have/What new integration you would like**
No problem, just figured out the OEM configuration for the new Sonoff Mini, at https://www.itead.cc/sonoff-mini.html

**Please describe your use case for this integration and alternatives you've tried:**
New Sonoff device, integrates a relay and a (user supplied) toggle switch.
Has the same out-of-the-box functionality as Sonoff Basic in a smaller package and with toggle switch screw terminals. First ESPHome firmware update OTA via Sonoff DIY tool.

**Additional context**
First firmware update OTA procedure described here: https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY [Flash procedure--Using the Itead DIY tool]

Example code:
'''
esphome:
  name: sonoff_mini
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_SSID
  password: !secret wifi_password

# Enable logging
logger:

substitutions:
  hostname: "sonoff_mini"

# Enable Home Assistant API
api:
  password: !secret default_password

ota:
  password: !secret default_password

# used just for testing, otherwise no need for it
#web_server:
#  port: 80

# tasmota template: {"NAME":"Sonoff Mini","GPIO":[17,0,0,0,9,0,0,0,21,56,0,0,255],"FLAG":0,"BASE":1}
# tasmota template decoded: GPIO: GPIO00 17 Button1, GPIO04 9 Switch1, GPIO12 21 Relay1, GPIO13 56 Led1i, GPIO16 255 userdef; BASE: Sonoff Basic

status_led:
  pin:
    number: GPIO13
    inverted: true

binary_sensor:
  - platform: gpio
    pin: GPIO00
    id: reset
    internal: true
    filters:
      - invert:
      - delayed_off: 10ms
    on_press:
      - switch.toggle: 
          id: relay_1

  - platform: gpio
    name: "$hostname switch"
    pin: GPIO04
    id: switch_1
    on_press:
      then:
        - switch.turn_on:
            id: relay_1
    on_release:
      then:
        - switch.turn_off:
            id: relay_1

switch:
  - platform: gpio
    name: "$hostname relay"
    icon: "mdi: lightbulb_outline"
    pin: GPIO12
    id: relay_1
    restore_mode: restore_default_off
'''

tortfeaser, 2019-08-16:
`esphome:
  name: sonoff_mini
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_SSID
  password: !secret wifi_password

# Enable logging
logger:

substitutions:
  hostname: "sonoff_mini"

# Enable Home Assistant API
api:
  password: !secret default_password

ota:
  password: !secret default_password

# used just for testing, otherwise no need for it
#web_server:
#  port: 80

# tasmota template: {"NAME":"Sonoff Mini","GPIO":[17,0,0,0,9,0,0,0,21,56,0,0,255],"FLAG":0,"BASE":1}

# template decoded: GPIO: GPIO00 17 Button1, GPIO04 9 Switch1, GPIO12 21 Relay1, GPIO13 56 Led1i, GPIO16 255 userdef; BASE: Sonoff Basic

status_led:
  pin:
    number: GPIO13
    inverted: true

binary_sensor:
  - platform: gpio
    pin: GPIO00
    id: reset
    internal: true
    filters:
      - invert:
      - delayed_off: 10ms
    on_press:
      - switch.toggle: 
          id: relay_1

  - platform: gpio
    name: "$hostname switch"
    pin: GPIO04
    id: switch_1
    on_press:
      then:
        - switch.turn_on:
            id: relay_1
    on_release:
      then:
        - switch.turn_off:
            id: relay_1

switch:
  - platform: gpio
    name: "$hostname relay"
    icon: "mdi: lightbulb_outline"
    pin: GPIO12
    id: relay_1
    restore_mode: restore_default_off`

OttoWinter, 2019-08-27:
Fixed in https://github.com/esphome/esphome-docs/commits/current

closed 2019-08-27 by OttoWinter