esphome/feature-requests #366: Possibility to fill a dark rectangle on a SSD1306 with a lux automation

Electronlibre2012, 2019-08-18:
Hello, i would like to use a photoresistance attached to the A0 of a ESP8266 to measure lux (already published by @OttoWinter ) and then fill a rectangle (like black screen) on a ssd1360.
The light of a SSD1360 is too bright for a room in dark when you want to sleep...
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

I try this code but dont know what to do after the "then"...how to interact with the ```- platform: ssd1306_i2c``` component, because he is in the ```display:``` and not in ```sensor:``` :

```
- platform: adc
    pin: A0
    name: Illuminance
    id: luminanceC1
    unit_of_measurement: lux
    filters:
      - lambda: return ( x * 3.3 / 10000.0) * 2000000.0 
    on_value:
      if:
        condition:
          sensor.in_range:
            id: my_sensor
            below: 10.0
        then:
          - script.execute: my_script 
```

**Additional context**

Electronlibre2012, 2019-08-18:
If someone is intersting one day... i found the solution 👍 

```
- platform: adc
    pin: A0
    name: Illuminance
    id: luminanceC1
    unit_of_measurement: lux
    filters:
      - lambda: return ( x * 3.3 / 10000.0) * 2000000.0;
```

```
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    lambda:  |-
      it.printf(30, 3, id(five8), "Capteurs Salon");
      it.printf(30, 3, id(five8), "___________");
      it.printf(0, 20, id(five8), "Temperature ......... %.1f °C", id(temperature).state);
      it.printf(0, 30, id(five8), "Humidite .............. %.1f RH", id(humidite).state);
      it.printf(0, 40, id(five8), "Pression ........... %.1f hPa", id(pressure).state);
      it.printf(0, 50, id(five8), "Luminosite........... %.1f lm", id(illumination_salon).state);
      it.printf(0, 60, id(five8), "Wifi : %s", id(etatW).state ? "Online" : "Offline");
      it.printf(85, 60, id(five8), "%.1f", id(signalW).state);
      it.printf(115, 60, id(five8), "dB");
      
      if (id(illumination_salon).state < 50) {
        it.fill(COLOR_OFF);
      } else {
      }
```

brandond, 2019-08-19:
Nice work! You might save yourself a few cycles and move all the printf stuff down into the else block so that you're not printing all that stuff just to blank it at the end.

Electronlibre2012, 2019-08-19:
yes! lol

thanks!!!

closed 2019-08-19 by brandond