esphome/feature-requests #368: Add delay timer for bang bang climate control

hpkfq, 2019-08-20:
**Describe the problem you have/What new integration you would like**

The current bang bang climate control is awesome, the easiest way to convert a relay to a wifi thermostat. However, there is a vital feature in short: delay timer.

According to this article:
https://highperformancehvac.com/delay-timers-air-conditioner-condenser-hvac-control/

It may be damageable to the compressor if it runs on and off frequently. Some of the compressors has built-in delay timers, some has not. But most of the digital thermostats have this function to protect compressor. It would be nice if this feature is added to the bang bang module. Thanks!
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

markabadee, 2019-08-29:
I would really like this feature. Not sure how I vote for this.

OttoWinter, 2019-08-29:
This is similar to `min_cycle_duration` for HA's generic thermostat: https://www.home-assistant.io/components/generic_thermostat/#min_cycle_duration

Yes it's not good if you turn certain devices on/off very often, though that's also pretty much exactly what bang_bang is for, by having two limits you essentially already have a hysteresis effect.

blitzaxt, 2020-01-02:
On a similar note is there a way to delay climate Bang-Bang after power loss. If not, this could probably help with that too.

bkaufx, 2021-06-06:
I'd like this feature as well.

nikolai5slo, 2022-03-16:
I would also like those `min_*` options that are in thermostat component also in bang-bang component.

closed 2022-07-01 by nagyrobi