esphome/feature-requests #369: Use light switch on wall to change functionality

shreddedbacon, 2019-08-20:
**Describe the problem you have/What new integration you would like**

It would be good to have the ability to flick the existing wall light switch quickly after booting to change functionality of the light, similar to how `on_boot` can set a default state, having something like `on_reboot` and having a different set of instructions run through

**Please describe your use case for this integration and alternatives you've tried:**

Use case would be something like:
Having an RGB light in the ceiling receptacle, and a normal wall switch, say in a rental where you can't touch the wiring.

At night time, flick the switch off then on again within say 1-2 seconds, and the light starts up in the second state on_reboot, which could be configurable to be like a nightlight mode.

In the morning, turn it off then back on and it boots back in the normal on_boot state which could be configured to be white light.

**Additional context**

Would expect yaml like the following
```
esphome:
  name: light-one
  platform: ESP8266
  board: esp01_1m
  on_boot:
    priority: 500
    then:
      - light.turn_on:
          <do_work>
on_reboot:
    priority: 500
    then:
      - light.turn_on:
          <do_different_work>
```

glmnet, 2019-08-21:
If I understand properly you want to power cycle the esp and then behave differently each time? 
That’s a bit different than reboot

shreddedbacon, 2019-08-21:
I guess it is a bit different to a reboot, but thats the best way I could think to word it.

But yes, the idea is that I can flick the wall switch twice to powercycle the device and it will start up in a different state.

Maybe it should be `on_powercycle` rather than `on_reboot`

shreddedbacon, 2019-08-21:
Ahh, I did not know about persistent variables!
I'll have a look and see if I can get it to work somehow using them.

shreddedbacon, 2019-08-21:
Tried to do a simple swap the bit on_boot
```
globals:
  - id: power_state
    type: bool
    restore_value: yes
    initial_value: 'true'
    
esphome:
  name: testlight
  platform: ESP8266
  board: esp01_1m
  on_boot:
    priority: 500
    then:
      - lambda: |-
          auto call = id(tlight).turn_on();
          if (id(power_state) == true) {
            call.set_brightness(0);
            call.set_white(0.85);
            id(power_state) = false;
          } else {
            call.set_white(0);
            call.set_effect("Rainbow");
            id(power_state) = true;
          }
          call.perform();
```
But it always starts up and the var never changes

I also tried
```
globals:
  - id: power_state
    type: bool
    restore_value: yes
    initial_value: 'true'
    
esphome:
  name: testlight
  platform: ESP8266
  board: esp01_1m
  on_boot:
    priority: 500
    then:
      - if:
          condition:
            lambda: 'return id(power_state);'
          then:
            - light.turn_on:
                id: tlight
                brightness: 0%
                white: 85%
            - globals.set:
                id: power_state
                value: 'false'
          else:
            - light.turn_on:
                id: tlight
                red: 10%
                green: 10%
                blue: 10%
                brightness: 10%
                white: 0%
                effect: "Rainbow"
            - globals.set:
                id: power_state
                value: 'true'
```
But it still never changes.

glmnet, 2019-08-21:
I'm sorry, I might confused what the `restore_value` does. There is a difference on keeping values on a reboot (software or reset push button, i.e. a reboot without losing power) and a full reboot of a power cycle.

The esp8266 has a limited flash write cycle count and is disabled by default. Try enabling the flash on the esp8266. See [this](https://esphome.io/components/esphome.html#esphome-esp8266-restore-from-flash)

shreddedbacon, 2019-08-21:
Aha! I think that did it.
Thanks for the pointers.

closed 2019-08-21 by shreddedbacon